smoltcp-rs/smoltcp #515: Ipv4Packet::new_checked always return truncated packet error

iamwwc, 2021-08-01:
[image]

#### How to reproduce it?

run test in follow git repo, there are always print `truncated packet
https://github.com/iamwwc/socks-tunnel/blob/master/src/tun/mod.rs

you may should set routes manually

https://github.com/iamwwc/socks-tunnel/blob/master/routes.sh

`/routes.sh ip 10.0.0.1 192.168.50.1 ens33 192.168.50.3`

change `192.168.50.1 ens33 192.168.50.3` to your interface address

Dirbaio, 2021-08-01:
Could you provide a standalone repro? Ideally a value of the `buffer &[u8]` that causes `new_checked` to return an error when it shouldn't.

iamwwc, 2021-08-01:
@Dirbaio https://github.com/iamwwc/smoltcp-maybe-bug try to run it :)

Dirbaio, 2021-08-01:
I meant if you could provide the value (bytes) of a packet that causes the issue. Anyway, checking your code, the packets look like this:

```
Length: 199 (0xc7) bytes
0000:   00 00 08 00  45 00 00 c3  72 d5 40 00  01 11 0c 5a   ....E...r.@....Z
0010:   0a 00 00 01  ef ff ff fa  de 51 07 6c  00 af 0b ac   .........Q.l....
0020:   4d 2d 53 45  41 52 43 48  20 2a 20 48  54 54 50 2f   M-SEARCH * HTTP/
0030:   31 2e 31 0d  0a 48 4f 53  54 3a 20 32  33 39 2e 32   1.1..HOST: 239.2
0040:   35 35 2e 32  35 35 2e 32  35 30 3a 31  39 30 30 0d   55.255.250:1900.
0050:   0a 4d 41 4e  3a 20 22 73  73 64 70 3a  64 69 73 63   .MAN: "ssdp:disc
0060:   6f 76 65 72  22 0d 0a 4d  58 3a 20 31  0d 0a 53 54   over"..MX: 1..ST
0070:   3a 20 75 72  6e 3a 64 69  61 6c 2d 6d  75 6c 74 69   : urn:dial-multi
0080:   73 63 72 65  65 6e 2d 6f  72 67 3a 73  65 72 76 69   screen-org:servi
0090:   63 65 3a 64  69 61 6c 3a  31 0d 0a 55  53 45 52 2d   ce:dial:1..USER-
00a0:   41 47 45 4e  54 3a 20 43  68 72 6f 6d  69 75 6d 2f   AGENT: Chromium/
00b0:   38 37 2e 30  2e 34 32 38  30 2e 31 34  31 20 4c 69   87.0.4280.141 Li
00c0:   6e 75 78 0d  0a 0d 0a                                nux....
truncated packet
```

The packet actually starts at `45 00 00 c3 ...`, the prepended 4 bytes are caused by the `config.packet_information(true);`. If you set it to `false` then it works fine:

```
Length: 195 (0xc3) bytes
0000:   45 00 00 c3  75 2d 40 00  01 11 0a 02  0a 00 00 01   E...u-@.........
0010:   ef ff ff fa  8a ed 07 6c  00 af 5f 10  4d 2d 53 45   .......l.._.M-SE
0020:   41 52 43 48  20 2a 20 48  54 54 50 2f  31 2e 31 0d   ARCH * HTTP/1.1.
0030:   0a 48 4f 53  54 3a 20 32  33 39 2e 32  35 35 2e 32   .HOST: 239.255.2
0040:   35 35 2e 32  35 30 3a 31  39 30 30 0d  0a 4d 41 4e   55.250:1900..MAN
0050:   3a 20 22 73  73 64 70 3a  64 69 73 63  6f 76 65 72   : "ssdp:discover
0060:   22 0d 0a 4d  58 3a 20 31  0d 0a 53 54  3a 20 75 72   "..MX: 1..ST: ur
0070:   6e 3a 64 69  61 6c 2d 6d  75 6c 74 69  73 63 72 65   n:dial-multiscre
0080:   65 6e 2d 6f  72 67 3a 73  65 72 76 69  63 65 3a 64   en-org:service:d
0090:   69 61 6c 3a  31 0d 0a 55  53 45 52 2d  41 47 45 4e   ial:1..USER-AGEN
00a0:   54 3a 20 43  68 72 6f 6d  69 75 6d 2f  38 37 2e 30   T: Chromium/87.0
00b0:   2e 34 32 38  30 2e 31 34  31 20 4c 69  6e 75 78 0d   .4280.141 Linux.
00c0:   0a 0d 0a                                             ...
239.255.255.250
```

closed 2021-08-01 by Dirbaio

iamwwc, 2021-08-02:
@Dirbaio sorry, my fault. Could you tell me how to print bytes like wireshark format. I should inspect my packet detail first

```
Length: 195 (0xc3) bytes
0000:   45 00 00 c3  75 2d 40 00  01 11 0a 02  0a 00 00 01   E...u-@.........
0010:   ef ff ff fa  8a ed 07 6c  00 af 5f 10  4d 2d 53 45   .......l.._.M-SE
0020:   41 52 43 48  20 2a 20 48  54 54 50 2f  31 2e 31 0d   ARCH * HTTP/1.1.
0030:   0a 48 4f 53  54 3a 20 32  33 39 2e 32  35 35 2e 32   .HOST: 239.255.2
0040:   35 35 2e 32  35 30 3a 31  39 30 30 0d  0a 4d 41 4e   55.250:1900..MAN
0050:   3a 20 22 73  73 64 70 3a  64 69 73 63  6f 76 65 72   : "ssdp:discover
0060:   22 0d 0a 4d  58 3a 20 31  0d 0a 53 54  3a 20 75 72   "..MX: 1..ST: ur
0070:   6e 3a 64 69  61 6c 2d 6d  75 6c 74 69  73 63 72 65   n:dial-multiscre
0080:   65 6e 2d 6f  72 67 3a 73  65 72 76 69  63 65 3a 64   en-org:service:d
0090:   69 61 6c 3a  31 0d 0a 55  53 45 52 2d  41 47 45 4e   ial:1..USER-AGEN
00a0:   54 3a 20 43  68 72 6f 6d  69 75 6d 2f  38 37 2e 30   T: Chromium/87.0
00b0:   2e 34 32 38  30 2e 31 34  31 20 4c 69  6e 75 78 0d   .4280.141 Linux.
00c0:   0a 0d 0a    
```

Dirbaio, 2021-08-02:
It's just `pretty_hex`: https://docs.rs/pretty-hex/0.2.1/pretty_hex/