smoltcp-rs/smoltcp #516: Exposing `process` and `accept` methods in TcpSocket and UdpSocket

zonyitoo, 2021-08-02:
We are trying to implement a proxy tunnel, which will automatically establish a new TCP connection or UDP association based on packet received from a `tun` device.

The current implementation requires `TcpSocket` to `listen`s or `UdpSocket` to `bind`s on a specific port otherwise the `interface` will ignore or send `RST` to TCP clients. But what I want is to create a new `TcpSocket` when receives a packet that points to a new destination and then proxy its packets to a proxy server.

It is required for us to use `process` to feed the received IP packets from `tun` devices to drive the TCP state forward:

https://github.com/smoltcp-rs/smoltcp/blob/1134eb28d8c6886034f6ccad452101dc5e6d679c/src/socket/tcp.rs#L1256-L1261

Dirbaio, 2021-08-02:
`process` and `accepts` on sockets are internal implementation details and are unlikely to be made public. They'd expose too much and make it harder to make changes.

There might be alternative ways? Maybe a way for TcpSocket/UdpSocket to listen on any ip+port, so Interface works for these transparent proxy use cases?

zonyitoo, 2021-08-02:
https://github.com/smoltcp-rs/smoltcp/blob/1134eb28d8c6886034f6ccad452101dc5e6d679c/src/iface/interface.rs#L650-L699

Since one call of `poll` may receives multiple IP packets, if these packets have different `dst_addr`s, then how can one `TcpSocket` or `UdpSocket` who listens on any ip+port handles all of them? Or I have to add many `TcpSocket`s into the socket set just in case?

zonyitoo, 2021-08-02:
How about making some new wrapper types, like `TcpRawSocket` and `UdpRawSocket`?

Dirbaio, 2021-08-02:
>  Or I have to add many TcpSockets into the socket set just in case?

Yes, that. You have to do that in regular TCP servers too in case you receive multiple connections. There's no "tcp listener" (yet?).

> How about making some new wrapper types, like TcpRawSocket and UdpRawSocket?

It has the same problems as exposing process/accepts: it exposes fairly low-level internal details in a public api, that then have to be maintained and compatibility kept. I'm not convinced that's the right approach, when Interface can be made to work.

Also it's not quite useful on its own, there's quite a bit of logic in Interface that you'd have to replicate...

zonyitoo, 2021-08-02:
If I added a lot of pre-generated `TcpSocket`s into the socket set, how can I identify which one just being connected? Or I have to maintain another set to record which sockets was in `open`ed state and then check it again after `poll`? With the `process` APIs, I can create one `TcpSocket` once I receive the first `SYN`, which is a lot graceful and elegant.

iamwwc, 2021-08-09:
Same here, I'm looking for tun implement (for shadowsocks-rust ;) ) using tcp/ip stack, but I found `tcp#process` function is not callable

leaf https://github.com/eycorsican/leaf/blob/b0779107921683204a65bb1d41edc07a52688613/leaf/src/proxy/tun/netstack/tcp_listener_impl.rs#L51 have a impl, which binding to a random port. 

What the leaf do is modifying `lwip source codes` to make all packets go through `the accept callback` regardless of tcp destination port.

https://github.com/eycorsican/lwip-leaf/search?q=TUN2SOCKS

https://github.com/eycorsican/lwip-leaf/blob/86632e2747c926a75d32be8bd9af059aa38ae75e/core/tcp_in.c#L686

Leaf's approach is good, but it need to hacking into library source codes, I want to find alternative approach to solve the problem.

As a tcp/ip stack, I think smoltcp should give developer more power to change/reroutes socket layer packet

blechschmidt, 2021-08-28:
While I agree that the ability to externally advance the socket state might be beneficial, one should already be able to implement a tool like tun2socks with zero modifications to smoltcp (master branch).

1. Instantiate a `TunTapInterface` for the device you want to operate on, e.g. tun0.
2. Implement a virtual device, i.e. implement the `Device` trait for a device that does nothing more except for providing the ability to manually add and remove packets to and from input and output buffers. The `receive` method will then read from the input buffer and `TxToken` consumption will write to the output buffer. Clone the capabilities from the `TunTapInterface`.
3. Instantiate an `Interface` for the virtual device with `any_ip`, route `0.0.0.1/0` and default gateway `0.0.0.1`. I will refer to this interface as virtual interface in the following.
4. Read from the `TunTapInterface` directly using `receive`. When you see a new TCP connection (SYN packet), instantiate a TcpSocket that listens on the destination endpoint and add it to a `SocketSet` you manage for the virtual interface. Then add the SYN packets and all following packets belonging to the connection to the input buffer of your virtual device.
5. Call `poll` on the virtual interface. It will read from the input buffer of the virtual device and will advance the states of the sockets in the `SocketSet`.
6. The previous call to `poll` may have caused the sockets to generate packets. Thus, read them from the output buffer of the virtual device.
7. Use `can_recv` and `recv` of the sockets normally.

Sending should work similarly.

zonyitoo, 2021-08-28:
It should be a practical solution. The key is to parse the received packets in the `Device` instance and check if there are new connections when receiving `SYN`s in `receive()`. I should try that.

blechschmidt, 2021-09-02:
I have been working on something similar. It's not complete yet, but if the proxy server does not misbehave, it works. Maybe this helps.

https://github.com/blechschmidt/tun2proxy

zonyitoo, 2022-01-15:
The method have been proved to work. Closing the issue.

closed 2022-01-15 by zonyitoo