smoltcp-rs/smoltcp #518: TCP socket begins to use improper window size

ryan-summers, 2021-08-12:
During analysis of wireshark traffic, it was noted that one of our TCP connections began using a window size much smaller than what the peer reported as available. Specifically in this case, the peer has a window size of 64KB, but smoltcp segments TCP messages into 63 bytes.

[image]

Above, you can see that at the start of the connection, smoltcp fits messages of >100 bytes into a single packet, as it notes that the window size is ~64KB.

However, then shortly later, a different socket (on the same interface) that was present on the peer, but is not registered in smoltcp (laying around from previous boot) gets cleaned up by the peer by sending a FIN-ACK. This socket had a window size of 63. Smoltcp then sends a RST to that connection, but then all subsequent communications on the new connection use a window size of 63 bytes instead of the correct window of ~64KB.

Dirbaio, 2021-08-12:
Interesting! Can you reproduce this every time? Can you post the .pcap and smoltcp verbose logs, if possible?

ryan-summers, 2021-08-12:
I only noticed this during a review of pcap files sent to me - I have not encountered this personally, so I do not know what the reproduction steps would be.

Here's the relevant section of the PCAP

[incorrect-window-size-adjustment.zip](https://github.com/smoltcp-rs/smoltcp/files/6975552/incorrect-window-size-adjustment.zip)

Dirbaio, 2021-08-12:
I see! Just to confirm, the MQTT payload in the last packet is incomplete, and communication stalls after that?

I guess then smoltcp waits forever for the ACK for the window to open up more, and the PC waits forever for the packet to complete? Why doesn't the PC ACK it? the packet looks valid.

Are you using master or 0.7.5?

`63` is the unscaled window length, so the bug is probably scaling not being applied correctly for whatever reason.

Dirbaio, 2021-08-12:
Maybe it doesn't ACK it because t

ryan-summers, 2021-08-12:
For clarity - this bug is not the connection stalling. This bug is purely that smoltcp began to segment up our TCP packets because it erroneously believed the window size was small.

The connection stalling"" issue I linked has a different root cause that is unrelated to smoltcp (the connection is fine, our scheduler is just wrong, so it wasn't sending anything out)

Dirbaio, 2021-08-12:
[are you](https://github.com/quartiq/stabilizer/blob/master/Cargo.lock#L779) on commit 027f255f904b9b7c4226cfd8b2d31f272ffa5105? there's been a few window scaling fixes since then: #500

ryan-summers, 2021-08-13:
Yes, we're likely tracking behind master due to the lockfile - we don't regularly update `smoltcp` - I'll try doing a dependency update and see if that resolves things. Thanks for the feedback!

jordens, 2021-08-18:
Ack. I think I'm seeing (in stabilizer-fls) that this is fixed in current smoltcp.

Dirbaio, 2021-09-08:
Closing as this seems fixed, feel free to open more issues if not.

closed 2021-09-08 by Dirbaio