smoltcp-rs/smoltcp #522: Splitting `src/iface/interface.rs` into multiple files

thibautvdv, 2021-08-19:
I was thinking of splitting the implementations of `Interface`, `InterfaceBuilder` and `InterfaceInner` into multiple files, where each file implements for a specific medium. I was working on the IEEE802.15.4 and 6LoWPAN protocol and I found that the `src/iface/interface.rs` was getting big with things that where sometimes unrelated. What do you think about this change?

But then again, I'm not sure if this is the best way to go as well, since, for example, `lookup_hardware_addr` is needed for Ethernet and IEEE802.15.4.

Dirbaio, 2021-09-08:
I did an attempt at splitting multiple mediums in multiple *structs* in #336 and it went really badly. `Interface` is really complex and Ethernet/IP aren't really independent of each other, so you can't simply split them like that. It was an uphill battle against the borrow checker :) 

What might work is splitting them in multiple *files*, but still being `impl`s on the same `Interface` struct. It's a mostly "cosmetic" improvement though, everything is still a megastruct, but it could still be worth it.

thibautvdv, 2021-09-09:
> What might work is splitting them in multiple files, but still being impls on the same Interface struct. It's a mostly "cosmetic" improvement though, everything is still a megastruct, but it could still be worth it.

I tried doing that as well and agree that it's very "cosmetic" and even worse than everything in one file. I'll close this issue then.

closed 2021-09-09 by thibautvdv