smoltcp-rs/smoltcp #527: Memory corruption: index out of range for slice of length 20

pmldrmota, 2021-09-11:
We are running smoltcp 0.6.0 on or1k (firmware equivalent to https://github.com/m-labs/artiq/tree/master/artiq/firmware/runtime) and are frequently (reproducibly, but not deterministically) seeing `panic at src/libcore/slice/mod.rs:1965:5: index 40 out of range for slice of length 20`, where length 20 seems to be constant and the index varies. I have very little expertise on this topic, so I am hoping that someone here can help. Below is the demangled backtrace. 
From this it seems that a TCP packet is dispatched, but then, an ICMPv4 packet is emitted - panicing when trying to set the echo ident. Is it possible that somehow, the socket buffer gets corrupted?
Is there a condition that facilitates this panic that we could try to prevent?

```
unwind_backtrace::backtrace at /artiq/firmware/libunwind_backtrace/lib.rs:42
 (inlined by) rust_begin_unwind at /artiq/firmware/runtime/main.rs:321
core::panicking::panic_fmt at /rust-core-or1k_1605281253004/work/src/libcore/panicking.rs:92
core::slice::slice_index_len_fail at /rust-core-or1k_1605281253004/work/src/libcore/slice/mod.rs:1965
<core::ops::range::Range<usize> as core::slice::SliceIndex<[T]>>::index_mut at /rust-core-or1k_1605281253004/work/src/libcore/slice/mod.rs:2142
 (inlined by) core::slice::<impl core::ops::index::IndexMut<I> for [T]>::index_mut at /rust-core-or1k_1605281253004/work/src/libcore/slice/mod.rs:1958
 (inlined by) <smoltcp::wire::icmpv4::Packet<T>>::set_echo_ident at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/wire/icmpv4.rs:327
 (inlined by) smoltcp::wire::icmpv4::Repr::emit at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/wire/icmpv4.rs:462
 (inlined by) smoltcp::iface::ethernet::InterfaceInner::dispatch::{{closure}} at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:1443
 (inlined by) smoltcp::iface::ethernet::InterfaceInner::dispatch_ip::{{closure}} at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:1681
 (inlined by) smoltcp::iface::ethernet::InterfaceInner::dispatch_ethernet::{{closure}} at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:1516
 (inlined by) <smoltcp::phy::tracer::TxToken<Tx, P> as smoltcp::phy::TxToken>::consume::{{closure}} at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/phy/tracer.rs:99
 (inlined by) <board_misoc::ethmac::EthernetTxSlot as smoltcp::phy::TxToken>::consume at /artiq/firmware/libboard_misoc/ethmac.rs:120
 (inlined by) <smoltcp::phy::tracer::TxToken<Tx, P> as smoltcp::phy::TxToken>::consume at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/phy/tracer.rs:98
 (inlined by) smoltcp::iface::ethernet::InterfaceInner::dispatch_ethernet at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:1511
 (inlined by) smoltcp::iface::ethernet::InterfaceInner::dispatch_ip at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:1668
 (inlined by) smoltcp::iface::ethernet::InterfaceInner::dispatch at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:1441

<smoltcp::iface::ethernet::Interface<'b, 'c, 'e, DeviceT>>::socket_egress::{{closure}} at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:595
 (inlined by) smoltcp::socket::tcp::TcpSocket::dispatch at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/socket/tcp.rs:1599
 (inlined by) <smoltcp::iface::ethernet::Interface<'b, 'c, 'e, DeviceT>>::socket_egress at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:625
 (inlined by) <smoltcp::iface::ethernet::Interface<'b, 'c, 'e, DeviceT>>::poll at /.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:495
 (inlined by) runtime::startup at /artiq/firmware/runtime/main.rs:220
core::ops::function::FnOnce::call_once at /rust-core-or1k_1605281253004/work/src/libcore/ops/function.rs:223
 (inlined by) logger_artiq::BufferLogger::register at /artiq/firmware/liblogger_artiq/lib.rs:67
 (inlined by) main at /artiq/firmware/runtime/main.rs:254
```

tballance, 2022-01-11:
We're hitting the same issue with a similar platform (artiq / or1k / smoltcp 0.6.0):

```
[2022-01-10 02:01:16.527301] panic at src/libcore/slice/mod.rs:1965:5: index 56 out of range for slice of length 20
[2022-01-10 02:01:16.527614] backtrace for software version 6.f18d38b9;primary:
[2022-01-10 02:01:16.559397] 0x4003226c /home/user/dev/artiq/artiq/firmware/libunwind_backtrace/lib.rs:42
[2022-01-10 02:01:16.559669] 0x4002341c /build/source/src/libcore/panicking.rs:92
[2022-01-10 02:01:16.575396] 0x40023b88 /build/source/src/libcore/slice/mod.rs:1965
[2022-01-10 02:01:16.591320] 0x400279e8 /build/source/src/libcore/slice/mod.rs:2142
[2022-01-10 02:01:16.591639] 0x400302e4 /home/user/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:595
[2022-01-10 02:01:16.607231] 0x400319ac /build/source/src/libcore/ops/function.rs:223
[2022-01-10 02:01:16.623219] restarting...
```

Did anyone figure this out?

adamgreig, 2022-01-17:
There's been [a ton of fixes](https://github.com/smoltcp-rs/smoltcp/blob/master/CHANGELOG.md) for problems found by fuzzing since 0.6.0 (which is two years old now), so I wouldn't be surprised if this was among them. Could you try with latest smoltcp 0.8? I don't think there's likely to be any backported fixes to 0.6 branch now...

It looks like there's an open PR for artiq that updates it to smoltcp 0.8, so that might be a convenient way to try.

Dirbaio, 2022-01-17:
The crashes look like memory corruption though. smoltcp has no `unsafe`, so it's unlikely that it's the cause of the corruption, it's probably just the "victim" of it.

Dirbaio, 2022-05-19:
As I said, smoltcp has no `unsafe` so it can't be *causing* corruption. The stack traces you're providing make no sense, as well. 

I'm pretty sure the issue is something else in the system corrupting memory used by smoltcp, causing it to panic.

`EthernetTracer` cannnot be causing the crashes, it's a noop device that passes buffers up and down unmodified, so it can't cause crashes in Interface. The fact the crashes go away if you remove it is further evidence that you have something else corruptiong smoltcp's memory: it moves the memory just enough that the corruption hits something else and no longer causes panics.

I suggest you update to 0.8 to benefit from the fixes. The stacktraces you're posting don't look like anything fixed by 0.7 or 0.8 though, so I'm not very hopeful. I suggest you look at the rest of your system for the corruption source.

closed 2022-05-19 by Dirbaio