smoltcp-rs/smoltcp #532: Support WiFi

naveedpash, 2021-09-21:
Is there any plan to support wireless interface?

Dirbaio, 2021-09-21:
Wifi is almost "Ethernet over the air". If you have a wifi chip that lets you send/receive ethernet frames, implementing `phy::Device` for it should work.

Actually implementing drivers for actual wifi chips, or the "below ethernet" layers like 802.11, WPA/WEP is out of scope. These drivers are usually closed-source blobs anyway...

thibautvdv, 2021-09-27:
I guess smoltcp could have another `Medium`,  like we have Ethernet and maybe in the future IEEE802.15.4 (#469) and maybe some day BLE. However, I don't think anybody is planning to add this to smoltcp.

MabezDev, 2022-03-29:
> Wifi is almost "Ethernet over the air". If you have a wifi chip that lets you send/receive ethernet frames, implementing phy::Device for it should work.

FYI we just did this with an esp32c3: https://github.com/bjoernQ/esp32c3-wifi-rs, currently PoC but will be expanded upon to support the rest of Espressif's chips.

Dirbaio, 2022-06-06:
Closing due to inactivity. If you have more questions, feel free to open new issues!

Dirbaio, 2022-06-06:
(tldr: wifi works with Medium::Ethernet, implementing lower-level 802.11 stuff is out of scope of smoltcp.)

closed 2022-06-06 by Dirbaio

pothos, 2022-06-06:
FYI: I've used smoltcp over WiFi both directly with macvtap in passthru mode as indirectly with usnetd driving the WiFi interface with macvtap in passthru mode and sharing it for multiple userspace network stacks and the kernel network stack:
https://github.com/ANLAB-KAIST/usnet_devices (for macvtap)
https://github.com/ANLAB-KAIST/usnet_sockets (high level socket library on top of smoltcp)
https://github.com/ANLAB-KAIST/usnetd

The code is not up to date and uses an old version of smoltcp, though.