smoltcp-rs/smoltcp #533: Crash due to assertion failure

black-binary, 2021-09-25:
My code crashed after inputting some packets. 

Here is a part of the stack backtrace:
```
Abort message: 'assertion failed: addr.is_unicast()'

(...)
core::panicking::panic::h945a80aaa5884f5b+48
smoltcp::iface::route::Routes::lookup::hc3ac269274d45dbb+1240
smoltcp::iface::interface::InterfaceInner::process_ip::hc48da65fc6377e67+1844
(my rx token consume function)
(interface::poll function)
(...)
```

`lookup` was called from here.

https://github.com/smoltcp-rs/smoltcp/blob/dedfbe1e7da2316b4e3c4a8c228c19b28190e3ea/src/iface/interface.rs#L1235-L1249

The assertion failed at the beginning of the `lookup` function and caused panic.

https://github.com/smoltcp-rs/smoltcp/blob/dedfbe1e7da2316b4e3c4a8c228c19b28190e3ea/src/iface/route.rs#L129-L133

black-binary, 2021-09-25:
Maybe we should fix this by adding more conditions.

https://github.com/smoltcp-rs/smoltcp/blob/dedfbe1e7da2316b4e3c4a8c228c19b28190e3ea/src/iface/interface.rs#L1235-L1237

->

```rust
 if !self.has_ip_addr(ipv4_repr.dst_addr) 
     && !self.has_multicast_group(ipv4_repr.dst_addr) 
     && !self.is_broadcast_v4(ipv4_repr.dst_addr) 
     && ipv4_repr.dst_addr.is_unicast()
```

Dirbaio, 2021-09-26:
Nice catch! The fix is adding the condition in the inner `if`, since we do want to ignore these packets (multicast dst_addr that we're not subscribed to). Fix in #534

commit 21e0a30d (2021-09-26): Fix assert with any_ip + broadcast dst_addr. Fixes #533
commit 2dc30c93 (2021-09-26): Merge #534

534: Fix assert with any_ip + broadcast dst_addr. r=Dirbaio a=Dirbaio

Fixes #533
commit db18514d (2021-09-26): Merge #534

534: Fix assert with any_ip + broadcast dst_addr. r=Dirbaio a=Dirbaio

Fixes #533
PR #534 (2021-09-26): Fix assert with any_ip + broadcast dst_addr.
Fixes #533
closed 2021-09-26 by bors[bot]