smoltcp-rs/smoltcp #537: arp: fill cache only from ARP responses

Dirbaio, 2021-10-01:
This fill here fills the cache on any ARP message. It should only do it on  ARP responses I think.

https://github.com/smoltcp-rs/smoltcp/blob/db18514d46d7c6c1b4305b7e4c5560372806cc84/src/iface/interface.rs#L1044

dnadlinger, 2021-10-01:
Isn't opportunistically use everything it sees to fill the cache a sensible thing to do?

Dirbaio, 2021-10-06:
@dnadlinger In theory it's correct, but I'm quite scared that some broken network out there breaks it, similarly to what happened in 6210612be047ee706ac729015cdbc2581e6ae9a3. Addresses in ARP packets should be more reliable yes, but still, who knows...

Also it can fill the the ARP cache with random entries we don't need, which can evict the ones that we do need. So this change can be good for performance too.

commit 83c01652 (2021-10-06): arp: fill cache only for ARP packets directed at us.

- Mirrors what Linux does, so will hopefully reduce problems in broken networks.
- it can actually increase performance: for small ARP caches, it'll reduce the
  amount of entries that we're not going to use, increasing the chances of the
  ones that we actually use to stay in the cache.

Fixes #537
closed 2021-10-06 by bors[bot]