smoltcp-rs/smoltcp #576: panicked at 'attempt to multiply with overflow', src/rand.rs:20:17

newAM, 2021-12-05:
With a debug build `rand_u32` panicked at this location:

https://github.com/smoltcp-rs/smoltcp/blob/774b375cb04e694199e27c7b9e36628436a4fac3/src/rand.rs#L20

That permalink contains the commit of smoltcp I am using (latest `master` when this issue was created).

I replaced it with this temporarily, works well enough for generating a DHCP XID.  I am not sure if this is the correct solution though :sweat_smile:

```rs
let s = self.state.overflowing_mul(M).0.wrapping_add(A);
```

Full backtrace:

<details>
<summary>Click to expand...</summary>
  
```
[INFO] Enable SRAM3
[INFO] Enabled cycle counter
[WARN] Checking clocks
[WARN] Clocks OK
[WARN] VERY SHORT DHCP LEASE TIME
[TRACE] [0]: adding
[WARN] Enabling ETH interrupt
[INFO] Done init
[INFO] idle
[INFO] Ethernet link status changed: false -> true
[INFO] ETH
[DEBUG] DHCP send DISCOVER to 255.255.255.255: Repr { message_type: Discover, transaction_id: 3, client_hardware_address: Address([2, 0, 0, 0, 0, 1]), client_ip: Address([0, 0, 0, 0]), your_ip: Address([0, 0, 0, 0]), server_ip: Address([0, 0, 0, 0]), router: None, subnet_mask: None, relay_agent_ip: Address([0, 0, 0, 0]), broadcast: false, requested_ip: None, client_identifier: Some(Address([2, 0, 0, 0, 0, 1])), server_identifier: None, parameter_request_list: Some([1, 3, 6]), dns_servers: None, max_size: Some(1432), lease_duration: None }
panicked at 'attempt to multiply with overflow', /home/alex/git/smoltcp/src/rand.rs:20:17
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: lib::inline::__udf
        at ./asm/inline.rs:172:5
   2: __udf
        at ./asm/lib.rs:49:17
   3: cortex_m::asm::udf
        at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/asm.rs:43:5
   4: rust_begin_unwind
        at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-probe-0.3.0/src/lib.rs:72:9
   5: core::panicking::panic_fmt
        at /rustc/f1edd0429582dd29cccacaf50fd134b05593bd9c/library/core/src/panicking.rs:100:14
   6: core::panicking::panic
        at /rustc/f1edd0429582dd29cccacaf50fd134b05593bd9c/library/core/src/panicking.rs:50:5
   7: smoltcp::rand::Rand::rand_u32
        at /home/alex/git/smoltcp/src/rand.rs:20:17
   8: smoltcp::socket::dhcpv4::Dhcpv4Socket::random_transaction_id
        at /home/alex/git/smoltcp/src/socket/dhcpv4.rs:374:9
   9: smoltcp::socket::dhcpv4::Dhcpv4Socket::dispatch
        at /home/alex/git/smoltcp/src/socket/dhcpv4.rs:400:35
  10: smoltcp::iface::interface::Interface<DeviceT>::socket_egress
        at /home/alex/git/smoltcp/src/iface/interface.rs:937:43
  11: smoltcp::iface::interface::Interface<DeviceT>::poll
        at /home/alex/git/smoltcp/src/iface/interface.rs:750:31
  12: cec::app::eth::{{closure}}
        at src/main.rs:306:29
  13: <(T1,T2) as rtic_core::prelude::TupleExt02>::lock::{{closure}}::{{closure}}
        at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/rtic-core-0.3.1/src/lib.rs:163:5
  14: rtic::export::lock
        at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtic-0.6.0-rc.4/src/export.rs:157:9
  15: cec::app::<impl rtic_core::Mutex for cec::app::shared_resources::dhcp>::lock
        at src/main.rs:85:1
  16: <(T1,T2) as rtic_core::prelude::TupleExt02>::lock::{{closure}}
        at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/rtic-core-0.3.1/src/lib.rs:124:26
  17: rtic::export::lock
        at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtic-0.6.0-rc.4/src/export.rs:157:9
  18: cec::app::<impl rtic_core::Mutex for cec::app::shared_resources::iface>::lock
        at src/main.rs:85:1
  19: <(T1,T2) as rtic_core::prelude::TupleExt02>::lock
        at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/rtic-core-0.3.1/src/lib.rs:155:21
  20: cec::app::eth
        at src/main.rs:303:9
  21: cec::app::ETH::{{closure}}
        at src/main.rs:85:1
  22: rtic::export::run
        at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtic-0.6.0-rc.4/src/export.rs:33:9
  23: ETH
        at src/main.rs:85:1
      <exception entry>
  24: <stm32h7xx_hal::ethernet::eth::EthernetMAC as stm32h7xx_hal::ethernet::StationManagement>::smi_read
        at /home/alex/git/stm32h7xx-hal/src/ethernet/eth.rs:727:15
  25: stm32h7xx_hal::ethernet::lan8742a::LAN8742A<MAC>::poll_link
        at /home/alex/git/stm32h7xx-hal/src/ethernet/lan8742a.rs:82:19
  26: cec::app::idle
        at src/main.rs:276:33
  27: main
        at src/main.rs:85:1
  28: Reset
(HOST) ERROR the program panicked
```

</details>

thibautvdv, 2021-12-06:
Indeed, overflows can happen, but I think you can use `wrapping_mul` instead of `overflowing_mul`:
```rust
let s = self.state.wrapping_mul(M).wrapping_add(A);
```

newAM, 2021-12-06:
> Indeed, overflows can happen, but I think you can use `wrapping_mul` instead of `overflowing_mul`:

🤦‍♂️ Not sure how I missed that method!  Thanks for the fix!

thibautvdv, 2021-12-06:
Thank you for noticing the overflow! :wink:

commit de109858 (2021-12-06): Merge #577

577: rand: fix overflow (issue #576) r=Dirbaio a=thibautvdv

Fixes #576
PR #577 (2021-12-06): rand: fix overflow (issue #576)
Fixes #576
closed 2021-12-06 by bors[bot]