smoltcp-rs/smoltcp #578: Branch medium-ip compile failed on linux

lazytiger, 2021-12-09:
I am using the medium-ip branch to implement a transparent proxy, and my application is fine on windows, but on Linux compiler complains about the following:

```rust
error[E0659]: `RawSocket` is ambiguous
   --> /home/hoping/.cargo/git/checkouts/smoltcp-d6a83fd2b0891566/0a1ad9e/src/iface/interface.rs:976:61
    |
976 |         for mut raw_socket in sockets.iter_mut().filter_map(RawSocket::downcast) {
    |                                                             ^^^^^^^^^ ambiguous name
    |
    = note: ambiguous because of multiple glob imports of a name in the same module
note: `RawSocket` could refer to the struct imported here
   --> /home/hoping/.cargo/git/checkouts/smoltcp-d6a83fd2b0891566/0a1ad9e/src/iface/interface.rs:11:5
    |
11  | use crate::phy::*;
    |     ^^^^^^^^^^^^^
    = help: consider adding an explicit import of `RawSocket` to disambiguate
note: `RawSocket` could also refer to the struct imported here
   --> /home/hoping/.cargo/git/checkouts/smoltcp-d6a83fd2b0891566/0a1ad9e/src/iface/interface.rs:14:5
    |
14  | use crate::socket::*;
    |     ^^^^^^^^^^^^^^^^
    = help: consider adding an explicit import of `RawSocket` to disambiguate

   Compiling webpki-roots v0.22.1
   Compiling rustls v0.20.2
error[E0599]: no function or associated item named `downcast` found for struct `phy::raw_socket::RawSocket` in the current scope
   --> /home/hoping/.cargo/git/checkouts/smoltcp-d6a83fd2b0891566/0a1ad9e/src/iface/interface.rs:976:72
    |
976 |         for mut raw_socket in sockets.iter_mut().filter_map(RawSocket::downcast) {
    |                                                                        ^^^^^^^^ function or associated item not found in `phy::raw_socket::RawSocket`
    |
   ::: /home/hoping/.cargo/git/checkouts/smoltcp-d6a83fd2b0891566/0a1ad9e/src/phy/raw_socket.rs:13:1
    |
13  | pub struct RawSocket {
    | -------------------- function or associated item `downcast` not found for this
    |
    = help: items from traits can only be used if the trait is implemented and in scope
note: `socket::AnySocket` defines an item `downcast`, perhaps you need to implement it
   --> /home/hoping/.cargo/git/checkouts/smoltcp-d6a83fd2b0891566/0a1ad9e/src/socket/mod.rs:157:1
    |
157 | pub trait AnySocket<'a, 'b>: SocketSession + Sized {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

```

Dirbaio, 2021-12-09:
The IP medium code is already merged to `master`, can you check if the issue is still present in `master`?

lazytiger, 2021-12-09:
Thanks for the reply, master is ok.

closed 2021-12-09 by lazytiger