smoltcp-rs/smoltcp #585: Add const generics to `Managed` classes

andrewgazelka, 2022-01-11:
What do you think of the idea of adding const generics to managed structs? For instance
```rust
enum ManagedMap<'a, K, V, const N: usize = 0> {
    Borrowed(&'a mut [Option<(K, V)>; N]),
    OwnedStd(BTreeMap<K, V>),
    OwnedCore([Option<(K, V)>; N]),
}
```

or if we want to make a more breaking change perhaps only allowing owned variants so we can reduce lifetime generics.

```rust
enum ManagedMap<K, V, const N: usize = 0> {
    OwnedStd(BTreeMap<K, V>),
    OwnedCore([Option<(K, V)>; N]),
}
```

Dirbaio, 2022-01-11:
The const generics would have to propagate through TcpSocket, UdpSocket, Socket, SocketSet, Interface, wouldn't they? I don't think 

IMO the current approach with slices is better, because
- It doesn't infect everything with const generics.
- Buffer size can be calculated at runtime
- Different sockets can have different buffer sizes.

andrewgazelka, 2022-01-11:
oh, that makes sense. I didn't factor that in. Thanks!

closed 2022-01-11 by andrewgazelka