smoltcp-rs/smoltcp #590: Code size benchmarks

fptech321, 2022-02-21:
Hi,

I haven't found any code size (flash and RAM) benchmarks for this library, which I think is very important for a library that is targeted for micro-controllers.

I conducted a small test where I compiled a minimal example with a udp socket (based on the examples [here](https://github.com/embassy-rs/ppproto)) and the code size (flash) I got was approximately **110 KB** which is quite huge (and not smol 😞 ) for a micro-controller, so I was wondering if I'm missing something, and if there are official benchmarks for the expected sizes?

The cargo target I used was `thumbv7m-none-eabi`.

crawford, 2022-02-21:
Are you building with `opt-level = "s"`?

fptech321, 2022-02-21:
yes, I've updated the original question.

adamgreig, 2022-02-21:
Can you post your example code anywhere? I wonder if your data buffers are ending up in flash memory (perhaps due to being part of a static that is not zero-initialised, so gets placed in `.data` instead of `.bss`). You could check using `cargo size --release -- -A` to see how big each section is.

Just to double check, are you also using `--release` when building, and measuring the size of the firmware in the release output `target/thumbv7m-none-eabi/release/`?

For comparison I have an entire application using TCP that's coming in at around 35kB so something is probably not quite right if you're getting 110kB for a minimal example!

crawford, 2022-02-21:
As an example, [my project](https://github.com/crawford/PoE/tree/13418fc6f6f43f26f53db4d2b0fde40c45895c34/firmware) uses TCP and DHCP sockets and weighs in at a little over 50 KiB:

```
$ cargo build --release --no-default-features --features silent
...
$ arm-none-eabi-size target/thumbv7m-none-eabi/release/poe -A
target/thumbv7m-none-eabi/release/poe  :
section             size        addr
.vector_table        336           0
.text              41788         336
.rodata             9644       42124
.data                948   536870912
.bss                5360   536871864
.uninit              112   536877224
.debug_abbrev       1722           0
.debug_info        13185           0
.debug_aranges       976           0
.debug_ranges       4800           0
.debug_str         26401           0
.debug_pubnames     8260           0
.debug_pubtypes      179           0
.ARM.attributes       48           0
.debug_frame        5124           0
.debug_line        14035           0
.debug_loc           109           0
.comment             109           0
Total             133136
```

fptech321, 2022-02-27:
Yep, my mistake. I haven't compiled it with `--release`. Thanks!

closed 2022-02-27 by fptech321