smoltcp-rs/smoltcp #592: Data not sent with TCP

hunhoffe, 2022-02-28:
Hello!

I am using smoltcp for a project and it was working great until I switched to using send_slice() and started sending multiple client messages at a time to a server. After those changes, sometimes the client believes data is sent to the server (the return of send_slice() indicates the data has been moved to the transmit buffer), but the data is never actually sent across the wire. 

I used send_queue() to check the amount of data in the tx_buffer, and the data appears to be there. I tried modifying my code to continuously poll the interface until the tx_buffer empties, but it never does.

I created a minimal example of the issue I see here: https://github.com/hunhoffe/smoltcptest

This could definitely be a user error but if it is, I'm not sure what I'm doing wrong. Any assistance or insight would be appreciated!

Dirbaio, 2022-02-28:
send_slice returns when the data has been enqueued in the tx buffer, that doesn't mean it's actually sent. Your client is enqueuing the data, then immediately exiting before it's sent.

Also, it is possible for send_slice/recv_slice to do "short writes": the returned value can be less than the buffer size, this meansonly part of the buffer has been sent/received and you have to try again later with the rest.

hunhoffe, 2022-02-28:
Hello! Thank you for your response. The example code in the repo I originally referenced does poll on the interface one additional time after the final call to send_slice() before exiting. However, for the sake of being removing any doubt, I've modified the code so it polls on the interface 100 times after the last call to send_slice(). The error does not change.  

You are correct - I forgot to check for "short writes"/"short reads" in my example. I've added those in and that is not the issue either. (My actual code is more robust than this and contain logic to ensure all data is sent/recv but I was trying to keep the code in the example minimal).

If you have any other ideas for what is causing this behavior, I'd appreciate your time!

Dirbaio, 2022-02-28:
It's not guaranteed that the next poll call sends the data. I think in this particular case it might be Nagle's Algorithm preventing it (there's already one smaller-than-MSS segment in flight, so it doesn't send another until the first is ACKed). There might be other causes, like slow network or peer reading data slowly causing the window to fill up.

In the general case if you want to make sure the peer gets the data, you have to make it respond with some "OK I got it" packet and wait for that.

hunhoffe, 2022-02-28:
I think your explanation makes sense, but then my question is: why isn't the server sending the ACK for the first packet? It's polling continually on the interface as it waits to receive the data, so I would have assumed smoltcp would send the ACK?

I thought TCP ACKs were supposed to ensure data is received at the peer, and that application logic should not be concerned with individual packet transmission. For context, I am implementing (a nearly) zero-copy RPC mechanism, so sometimes one application message is composed of multiple calls to send_slice() on several different client buffers. However, the server only responds when the whole message has been received.

Dirbaio, 2022-02-28:
> why isn't the server sending the ACK for the first packet?

From the tcpdump in your readme it seems it does, but I guess your client binary has already exited by then.

> I thought TCP ACKs were supposed to ensure data is received at the peer, and that application logic should not be concerned with individual packet transmission.

It does indeed. As I said, the problem is your program is not waiting for the ACK, it's pushing the data to the buffer then immediately exiting.

hunhoffe, 2022-02-28:
I see! Yes, thank you for explaining. I appreciate your time!

gopakumarce, 2022-02-28:
@hunhoffe is there any iptables/rpf_forwarding something I have to tweak with to get your example working ? The Syn from tap2 never gets to tap0 - the mac address / arp tables are populated all right.

Dirbaio, 2022-02-28:
You need to pass the real time to .poll(): https://github.com/hunhoffe/smoltcptest/blob/less-minimal/src/common.rs#L171

smoltcp is probably waiting for some timer to expire, if you pass Instant(0) it'll wait forever.

hunhoffe, 2022-02-28:
@Dirbaio  It was the timer issue! Thanks much!

closed 2022-02-28 by hunhoffe