smoltcp-rs/smoltcp #595: bpf.rs build failed on iOS

zonyitoo, 2022-03-29:
```
cargo build --target aarch64-apple-ios --features "local-tun local-redir armv8 neon"
   Compiling shadowsocks-service v1.13.5 (/Users/xiewenyu/Desktop/rust-project/shadowsocks-rust/crates/shadowsocks-service)
   Compiling shadowsocks-rust v1.13.5 (/Users/xiewenyu/Desktop/rust-project/shadowsocks-rust)
    Finished dev [unoptimized + debuginfo] target(s) in 19.23s
xiewenyudeMBP:shadowsocks-rust xiewenyu$ cargo build --target aarch64-apple-ios --features "local-tun local-redir armv8 neon"
   Compiling smoltcp v0.8.0
   Compiling shadowsocks v1.13.2 (/Users/xiewenyu/Desktop/rust-project/shadowsocks-rust/crates/shadowsocks)
error[E0425]: cannot find value `BIOCSETIF` in this scope
  --> /Users/xiewenyu/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.0/src/phy/sys/bpf.rs:79:29
   |
79 |         try_ioctl!(self.fd, BIOCSETIF, &mut self.ifreq);
   |                             ^^^^^^^^^ not found in this scope
   |
help: consider importing one of these items
   |
1  | use crate::phy::sys::bpf::libc::BIOCSETIF;
   |
1  | use libc::BIOCSETIF;
   |

error[E0425]: cannot find value `BIOCIMMEDIATE` in this scope
   --> /Users/xiewenyu/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.0/src/phy/sys/bpf.rs:107:29
    |
107 |         try_ioctl!(self.fd, BIOCIMMEDIATE, &mut bufsize as *mut libc::c_int);
    |                             ^^^^^^^^^^^^^ not found in this scope
    |
help: consider importing one of these items
    |
1   | use crate::phy::sys::bpf::libc::BIOCIMMEDIATE;
    |
1   | use libc::BIOCIMMEDIATE;
    |

error[E0425]: cannot find value `BIOCGBLEN` in this scope
   --> /Users/xiewenyu/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.0/src/phy/sys/bpf.rs:108:29
    |
108 |         try_ioctl!(self.fd, BIOCGBLEN, &mut bufsize as *mut libc::c_int);
    |                             ^^^^^^^^^ not found in this scope
    |
help: consider importing one of these items
    |
1   | use crate::phy::sys::bpf::libc::BIOCGBLEN;
    |
1   | use libc::BIOCGBLEN;
    |

error[E0425]: cannot find value `BPF_HDRLEN` in this scope
   --> /Users/xiewenyu/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.0/src/phy/sys/bpf.rs:121:35
    |
121 |             if len == -1 || len < BPF_HDRLEN as isize {
    |                                   ^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `BPF_HDRLEN` in this scope
   --> /Users/xiewenyu/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.0/src/phy/sys/bpf.rs:129:25
    |
129 |                 &buffer[BPF_HDRLEN] as *const u8 as *const libc::c_void,
    |                         ^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `BPF_HDRLEN` in this scope
   --> /Users/xiewenyu/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.8.0/src/phy/sys/bpf.rs:130:23
    |
130 |                 len - BPF_HDRLEN,
    |                       ^^^^^^^^^^ not found in this scope

For more information about this error, try `rustc --explain E0425`.
error: could not compile `smoltcp` due to 6 previous errors
warning: build failed, waiting for other jobs to finish...
error: build failed.
```

Original issue: https://github.com/shadowsocks/shadowsocks-rust/issues/797#issuecomment-1081339878

It seems that some constants, like `BIOCSETIF`, `BPF_HDRLEN` is only available for `#[cfg(any(target_os = "macos", target_os = "openbsd"))]`.

zonyitoo, 2022-03-29:
These lines in `src/phy/sys/mod.rs` should be changed to:

```rust
#[cfg(all(
    feature = "phy-raw_socket",
    any(target_os = "macos", target_os = "openbsd"),
    unix
))]
pub mod bpf;

#[cfg(all(
    feature = "phy-raw_socket",
    any(target_os = "macos", target_os = "openbsd"),
    unix
))]
pub use self::bpf::BpfDevice as RawSocketDesc;
```

623637646, 2023-01-13:
I use this temporary solution to fix my issues. 
Change `smoltcp = "0.8.2"` 
to `smoltcp = { version = "0.8.2", default-features = false, features = ["proto-ipv4"] }` 
in the Cargo.toml