smoltcp-rs/smoltcp #607: tcpsocket coupled to interface

dvc94ch, 2022-05-01:
it seems like process and dispatch are used to feed tcp packets into the state machine. in my case there is a usb bulk endpoint that has a simple protocol to send tcp packets. however it doesn't seem like there is any public api to get tcp packets out of the TcpSocket so I can send them via the usb endpoint.

- [0] https://docs.rs/smoltcp/latest/smoltcp/socket/struct.TcpSocket.html

Dirbaio, 2022-05-01:
smoltcp is designed to be used as a full network stack. There's no support for using "just" TCP, as that is very rare and would require a lot of extra API surface.

What do you need exactly? Packets with TCP headers but no IP/Ethernet headers? That's a very strange thing, what device is it?

dvc94ch, 2022-05-01:
iphones use the usbmux protocol to transfer tcp packets.

> There is no reordering or retransmission. There are no checksums, no URG, no
PSH, no non-ACK, no FIN. What there *is* is the SEQ/ACK/window mechanism used
for flow control, and RST is used as the only connection teardown mechanism (and
also for "connection refused"), and the connection startup is SYN/SYNACK/ACK.

reading more about this, it looks like I'll have to implement something specifically for usbmuxd.

closed 2022-05-01 by dvc94ch