smoltcp-rs/smoltcp #613: ICMPv4 outgoing source address missing

bunnie, 2022-05-12:
In the latest version of smoltcp there is an assert on line 2420 of
interfaec.rs which requires that ip_repr.src_adr().is_unspecified() be
false: https://github.com/smoltcp-rs/smoltcp/blob/007f8dd2e2d0a770609428aa4b0efb63f5dbb708/src/iface/interface.rs#L2420

This contradicts the construction of the icmp v4 packet in the dispatch()
function of icmp.rs, where the src_addr is set to an unspecified default: https://github.com/smoltcp-rs/smoltcp/blob/007f8dd2e2d0a770609428aa4b0efb63f5dbb708/src/socket/icmp.rs#L454

I opened a PoC PR of a fix at #612. This basically adds a function to specify an IPv4 source and wedge it into the packet. It's not the "right" fix, but I can confirm that with this in place, I no longer trip the assert and pings work correctly.

Dirbaio, 2022-05-12:
root cause is #579 , same as #599 .

The bug is not present in the released 0.8.0, so I suggest you use that in the meantime. There's no significant improvements in git master since 0.8.0.

bunnie, 2022-05-12:
Ah thanks, I see! Actually, the reason we couldn't use 0.8.0 release is 4a489b38db513246c3d2ea722b38fa1ae767a6ad -- we need to have the rand_core dependency removed because with stock 0.8.0 we get this error as we have a custom OS target:
```
error: target is not supported, for more information see: https://docs.rs/getrandom/#unsupported-targets
   --> C:\Users\bunnie\.cargo\registry\src\github.com-1ecc6299db9ec823\getrandom-0.2.5\src\lib.rs:234:9
    |
234 | /         compile_error!("target is not supported, for more information see: \
235 | |                         https://docs.rs/getrandom/#unsupported-targets");
    | |________________________________________________________________________^

error[E0433]: failed to resolve: use of undeclared crate or module `imp`
   --> C:\Users\bunnie\.cargo\registry\src\github.com-1ecc6299db9ec823\getrandom-0.2.5\src\lib.rs:256:5
    |
256 |     imp::getrandom_inner(dest)
    |     ^^^ use of undeclared crate or module `imp`
```
Commits later than that work, so maybe we can target d05ebdd88f7f2b87a2ffe3256a42f67809d2d7e4...

Thanks for the fast feedback, tho!

Dirbaio, 2022-05-12:
oh, I didn't realize `rand_core` would be that troublesome! Released 0.8.1 removing it (and some minor fixes).

bunnie, 2022-05-13:
0.8.1 works for us! Feel free to close this if you think the other bugs track this well enough, or leave it open if you want to be reminded of the test case.

thanks for the very speedy turn-around. I really appreciate it.