smoltcp-rs/smoltcp #621: Create a `sans-io`-style example

KOLANICH, 2022-05-29:
Hi. I currently need a permissively licensed userspace [`sans-io`](https://sans-io.readthedocs.io/)-style TCP/IP stack, because my app is not going to deal with raw sockets at all, instead it will en/decapsulate IP packets into/from UDP packets of WireGuard protocol (BoringTun impl, also supports usage in sans-io fashion and exposes the API in the form of functions callable from C, so I have written the python bindings for it). So I wonder if this library supports usage in `sans-io` fashion, and if it does, could you please provide an example?

Dirbaio, 2022-05-29:
You can supply a `phy::Device` that does whatever you want, including "fake" IO like pushing the packets to a vec/queue and then having other code do stuff with it. There's examples of `phy::Device` impls inside the `phy` module.

closed 2022-05-29 by KOLANICH

KOLANICH, 2022-08-31:
Here it is: https://github.com/KOLANICH-libs/sio-smoltcp.rs .