smoltcp-rs/smoltcp #629: Site-local 6LowPan

benbrittain, 2022-06-12:
I'm trying to get two microcontrollers to talk to each other over 805.15.4, both assigned static site-local IP addresses derived from the device's hardware address. This appropriately triggers a ICMPv6 NDISC request, but after that things look a little weird to me.

The ICMPv6 type being output is 0, which I don't think is a valid type. This is from the `c4791eb668c421f941c4c489ba1a402162b7be93` commit of upstream.

```TRACE - udp:fd00:1eaf::a08d:cfd3:fffe:bde1:7:fd00:1eaf::5ad3:fcc:fffe:7391:7: buffer to send 2 octets
TRACE - udp:fd00:1eaf::a08d:cfd3:fffe:bde1:7:fd00:1eaf::5ad3:fcc:fffe:7391:7: sending 2 octets
DEBUG - address fd00:1eaf::5ad3:fcc:fffe:7391 not in neighbor cache, sending Neighbor Solicitation```

Here is the PCAP file from the sending side, renamed to a txt for github upload. You might need to disable the "Disect only good FCS" in the 802.15.4 protocol preferences since my radio peripheral calculates that after this is dumped.
[gateway.txt](https://github.com/smoltcp-rs/smoltcp/files/8886980/gateway.txt)

benbrittain, 2022-06-12:
@thvdveld who has some context on this

thvdveld, 2022-06-13:
I don't see how `smoltcp` can set all fields of the ICMPv6 packet, but not the type.
I think I'll require some help from @Dirbaio.

benbrittain, 2022-06-13:
If you look at the bytes in the pcap file (and I confirmed with some printf debugging), It does call the set_type path, it's just not in the right place? I'm still rather fuzzy on how the compression iphc header stuff works, but you can see the `87` bit on line `0020`

```
0000   41 c8 03 af 1e ff ff e1 bd fe ff d3 cf 8d a0 7b   A..............{
0010   39 3a fd 00 1e af 00 00 00 00 a0 8d cf d3 ff fe   9:..............
0020   bd e1 02 01 ff fe 73 91 87 00 92 37 00 00 00 00   ......s....7....
0030   fd 00 1e af 00 00 00 00 5a d3 0f cc ff fe 73 91   ........Z.....s.
0040   01 02 a0 8d cf d3 ff fe bd e1 00 00 00 00 00 00   ................
```

thvdveld, 2022-06-13:
Could you share a link to your repository? I wrote a test for it and from the test it seems to work like expected.

benbrittain, 2022-06-13:
The repository is a bit gnarly at the moment and depends on a bunch of local crates, but here is the setup code. The issue shouldn't be with the mcu radio stuff since this is dumped from smoltcp before we ever get to the write path of the radio peripheral.

https://gist.github.com/benbrittain/b10c228e369b19c7185a80750dc735fd

benbrittain, 2022-06-13:
the IPv6 site-local address being generated for the radio peripheral is  `fd00:1eaf::a08d:cfd3:fffe:bde1/64`

thvdveld, 2022-06-13:
Now my test fails indeed, I will try to figure out why it fails now!

Dirbaio, 2022-06-13:
reopening pending confirmation it fixes OP's bug

benbrittain, 2022-06-13:
wow, so speedy! Setting the microcontrollers back up (which I tore apart to test something else because I wasn't expecting this so quickly :sweat_smile: )

I do see something a little weird with the dumped packets in wireshark on the sending side though, it looks like maybe the checksum isn't being generated quite right?

[image]

commit 46012c86 (2022-06-13): Merge #630

630: Fix 6LoWPAN address compression r=Dirbaio a=thvdveld

Should fix #629. The `Source Address Mode (SAM)` in the 6LoWPAN `IPHC` header was set to `Compress` (`0b11`), while it should have been `Inline` (`0b00`) when we inline the full IPv6 address.
commit 3e68df92 (2022-06-13): Merge #630

630: Fix 6LoWPAN address compression r=Dirbaio a=thvdveld

Should fix #629. The `Source Address Mode (SAM)` in the 6LoWPAN `IPHC` header was set to `Compress` (`0b11`), while it should have been `Inline` (`0b00`) when we inline the full IPv6 address.
PR #630 (2022-06-13): Fix 6LoWPAN address compression
Should fix #629. The `Source Address Mode (SAM)` in the 6LoWPAN `IPHC` header was set to `Compress` (`0b11`), while it should have been `Inline` (`0b00`) when we inline the full IPv6 address.