smoltcp-rs/smoltcp #647: 6LowPan packet fragmentation buffer logic

benbrittain, 2022-08-03:
Setting up the packet assembler isn't as clear as I'd assumed.

My assumption was that the individual parts of the fragments cache would be the size of the MTU of the 802.15.4 L2 layer (127). Since the MAC payload can be as few as 74 bytes, we'd need at least 18 fragments worth of buffer in the `PacketAssembler` to reassemble a full UDP frame.

a packet buffer looking like: `[[u8; 127]; 18]`, doesn't seem to work with the current implementation however.

The logic inside the fragmentation module looks to assume that the 0'th buffer is the full size of the reassembled frame.

```
    pub(crate) fn start(                                    
        &mut self,                                          
        total_size: Option<usize>,                          
        expires_at: Instant,                                
        offset_correction: isize,                           
    ) -> Result<()> {                                       
        match &mut self.buffer {                            
            ManagedSlice::Borrowed(b) => {                  
                if let Some(total_size) = total_size {      
                    if b.len() < total_size {               
                        return Err(Error::PacketAssemblerBufferTooSmall);                                               
                    }                                       
                }                                           
            }    
```

If I try to send a that is larger than an individual fragment buffer, I get an error that looks like

```
TRACE - iface: malformed check!(fragments.reserve_with_key(&
                                  key)).start(Some(frag.datagram_size() as
                                                       usize -
                                                       uncompressed_header_size
                                                       +
                                                       compressed_header_size),
                                              self.now +
                                                  Duration::from_secs(60),
                                              -((uncompressed_header_size -
                                                     compressed_header_size)
                                                    as isize))                                                
```

If I increase the size of the FIRST individual fragment buffer in the packet assemble cache, I'm able to receive more fragments (and reassemble them!), but I can even set the 1..N fragment buffers to a 0 sized array and still assemble at least some packets.

Should I be setting up these buffer differently or is there an issue here?

benbrittain, 2022-08-03:
@thvdveld :)

thvdveld, 2022-08-04:
Reassembly in `smoltcp` has two structs:
- `PacketAssemblerSet`: this struct contains multiple `PacketAssembler`s (for example when different streams of fragments arrive, they all need to be assembled with their own `PacketAssembler`).
- `PacketAssembler`: this is for assembling just one stream of fragments. It does this by using a big chunk of memory (and not different chunks that have the size of the MTU of IEEE802.15.4). It also has an `AssemblerState` which holds information about the total length and the places which already have been filled with received fragments in the big memory chunk (and other information that is needed for reassembly). So the buffer that you would use for the `PacketAssembler` is just one big array, which ideally should have at least around the size of the MTU of IPv6.

benbrittain, 2022-08-04:
That all makes sense, turns out part of my confusion was derived from the radio I'm working with being a bit lossy and very highly fragmented packets rarely arrive in their totality. I added a little bit of info in #648 and #649

closed 2022-08-04 by benbrittain