smoltcp-rs/smoltcp #654: Fragmentation Interface builder panics

Dirbaio, 2022-08-06:
When enabling fragmentation-related features, the builder requires passing in the corresponding fragmentation buffers. Otherwise `finalize` panics.

Ideally, enabling a Cargo feature shouldn't break working code: features should be additive.

The fix should be using zero-length buffers if we're not given any (so that fragmentation/reassembly won't happen, but previously-working code keeps working)

thvdveld, 2022-08-16:
The thing is that the buffers are `Into<ManagedMap>` or `Into<ManagedSlice>`. And these are only implemented for slices and not for arrays, because arrays need to be boxed. So I'm not sure how we would create a buffer (if using `Managed`, because it doesn't allow us).

Also, the reason I made it panicking is that I looked at the `neighbour_cache` for Ethernet. But I understand that indeed it is different from Ethernet because the medium is not an option, it's something we really have to specify. I'll try to change it, but I'm struggling because of `Managed`.

Dirbaio, 2022-08-16:
zero-length slice literals (&mut []) don't need to be allocated "anywhere" so they can be used as defaults.

thvdveld, 2022-08-16:
Ah right, I just had to write `&mut [][..]` instead of the `&mut []` that I tried :sweat_smile:

thvdveld, 2022-09-28:
This is fixed in #634

closed 2022-09-28 by thvdveld