smoltcp-rs/smoltcp #655: Creating assemblers in nostd is quite hard.

Dirbaio, 2022-08-06:
Currently you have to do something like this

```rust
    let mut ipv4_in1 = [0u8; 1500];
    let mut ipv4_in2 = [0u8; 1500];
    let mut ipv4_in3 = [0u8; 1500];
    let mut ipv4_in4 = [0u8; 1500];
    let mut ipv4_fragments_assemblers = [
        PacketAssembler::new(&mut ipv4_in1[..]),
        PacketAssembler::new(&mut ipv4_in2[..]),
        PacketAssembler::new(&mut ipv4_in3[..]),
        PacketAssembler::new(&mut ipv4_in4[..]),
    ];
    let mut ipv4_fragments_indexes = [None; 4];
    let ipv4_fragments_cache = FragmentsCache::new(
        &mut ipv4_fragments_assemblers[..],
        &mut ipv4_fragments_indexes[..],
    );
    buider = builder.ipv4_fragments_cache(ipv4_fragments_cache);
```

(unless I'm missing something?)

thvdveld, 2022-08-16:
That is the correct way of creating the buffers. Unfortunately this is because of `Managed`, and I'm not sure how to improve on that.

Dirbaio, 2023-01-15:
Fixed in #726

closed 2023-01-15 by Dirbaio