smoltcp-rs/smoltcp #694: is this a bug?

rongyi, 2022-10-14:
https://github.com/smoltcp-rs/smoltcp/blob/master/src/storage/assembler.rs#L210

 let mut assr = contigs![(0, 8), (4, 4)];
        println!("here");
        assert_eq!(assr.add(8, 4), Ok(false));

this testcase will panic at this line. I was just read smoltcp code, maybe the input is not valid, please correct me if I'm wrong.

Dirbaio, 2022-10-23:
indeed it is...

another repro:

```rust

    #[test]
    fn test_lol() {
        let mut assr = Assembler::new(16);
        assert_eq!(assr.add(0, 1), Ok(false));
        assert_eq!(assr.add(2, 1), Ok(false));
        assert_eq!(assr.add(1, 1), Ok(false));
    }
```

closed 2022-10-23 by bors[bot]

Dirbaio, 2022-10-23:
Thanks for finding and reporting it! :)

rongyi, 2022-11-07:
@Dirbaio pleasure!

commit a55c00fa (2022-10-23): assembler: do not return whether there was an overlap.

This functionality is completely unused, and was panicky with overflow checking (see #694)
Fixes #694
commit 6c2352be (2022-10-23): Merge #700

700: assembler: do not return whether there was an overlap. r=Dirbaio a=Dirbaio

This functionality is completely unused, and was panicky with overflow checking (see #694).

Fixes #694

bors r+
PR #700 (2022-10-23): assembler: do not return whether there was an overlap.
This functionality is completely unused, and was panicky with overflow checking (see #694).

Fixes #694

bors r+
commit 11b93261 (2022-11-14): assembler: do not return whether there was an overlap.

This functionality is completely unused, and was panicky with overflow checking (see #694)
Fixes #694