smoltcp-rs/smoltcp #702: Problems using `AsRef` and `AsMut`

thvdveld, 2022-10-27:
For the moment, all `Packet` representations in `src/wire/*` have the following definition:

```rust
pub struct Packet<T: AsRef<[u8]>> {
    buffer: T,
}

impl<'p, T: AsRef<[u8]>> Packet<&'p T> {
    fn get_some_data(&self) -> &'p [u8] {
        &self.buffer.as_ref()[1..]
    }
}
```

Because of the variance rules, the lifetime of `as_ref()` has a lifetime of `'p`, instead of the lifetime of `&self`. However, when we want to have a mutable reference to some data:

```rust
impl<'p, T: AsRef<[u8]> + AsMut<[u8]>> Packet<&'p T> {
    fn get_some_data_mut(&mut self) -> &'p mut [u8] {
        &mut self.buffer.as_mut()[1..]
    }
}
```

The variance rules change and the lifetime of `as_mut()` cannot be `'p`, but needs to be the same as `&mut self`. In some cases, this can be annoying when implementing parsers and emitters.

---

The only fix that I have for this issue is removing the generic argument and create two type of packets:

```rust
struct Packet<'p> {
    buffer: &'p [u8],
}

struct PacketMut<'p> {
    buffer: &'p mut [u8],
}
```
The getters and setters would then be:

```rust
impl<'p> Packet<'p> {
    fn get_some_data(&self) -> &'p [u8] {
        &self.buffer[1..]
    }
}

impl<'p> PacketMut<'p> {
    fn into_ref(&self) -> Packet<'p> {
        Packet { buffer: &*self.buffer }
    } 

    fn get_some_data(&mut self) -> &'p mut [u8] {
        &mut self.buffer[1..]
    }
}
```

This solves the lifetime issues and even makes the code simpler. The only downside of this is that now we need to call `packet.into_ref()` to be able to use the immutable getters.

I already played with this in my `rpl-wire` branch, but I would like to know what you all think about this.

Dirbaio, 2022-10-27:
I'm not sure if this is something we can fix, it's a direct consequence of the Rust borrow rules. You can have multiple `&` XOR single `&mut` pointing to some data at a time.

`fn get_some_data_mut(&mut self) -> &'p mut [u8]` can't work because it borrows `self` for too little time, you could call it twice and get two `&'p mut`s pointing to the same data. FWIW your propsed alternative doesn't compile either [plaground](https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=d9d78421205cff23cee9014a28b4f24c).

It is annoying if you want to get `&mut`s to two fields in the same packet at once, yes... 

You could do it if packets were `repr(C)` structs, because the borrow checker is smart enough to "see" you're only borrowing one field and not the whole thing, but it's not feasible to represent all packets with `repr(C)`, especially stuff with variable lengths or optional fields...

thvdveld, 2022-10-28:
Sorry, I think I found where my issue was.

closed 2022-10-28 by thvdveld