smoltcp-rs/smoltcp #715: defmt does not work with alloc feature

stevefan1999-personal, 2022-12-10:
This error is pretty curious, I wonder if that is due to `core` changes since I remember `Box` should have had handled `Format` to passthrough transparently to the underlying type:
```
(base) PS C:\Users\steve\esp32c2-rust-hello-world> cargo build
   Compiling smoltcp v0.8.2
   Compiling esp32c3-hal v0.3.0
   Compiling esp-wifi v0.1.0 (https://github.com/esp-rs/esp-wifi.git?rev=4d8243f#4d8243f7)
error[E0277]: the trait bound `Box<[Contig; 32]>: Format` is not satisfied
   --> C:\Users\steve\.cargo\registry\src\github.com-1ecc6299db9ec823\smoltcp-0.8.2\src\storage\assembler.rs:104:5
    |
99  | #[cfg_attr(feature = "defmt", derive(defmt::Format))]
    |                                      ------------- required by a bound introduced by this call
...
104 |     contigs: Box<[Contig; CONTIG_COUNT]>,
    |     ^^^^^^^ the trait `Format` is not implemented for `Box<[Contig; 32]>`
    |
    = help: the following other types implement trait `Format`:
              &T
              &mut T
              ()
              (T0, T1)
              (T0, T1, T2)
              (T0, T1, T2, T3)
              (T0, T1, T2, T3, T4)
              (T0, T1, T2, T3, T4, T5)
            and 185 others
note: required by a bound in `defmt::export::fmt`
   --> C:\Users\steve\.cargo\registry\src\github.com-1ecc6299db9ec823\defmt-0.3.2\src\export\mod.rs:137:15
    |
137 | pub fn fmt<T: Format + ?Sized>(f: &T) {
    |               ^^^^^^ required by this bound in `defmt::export::fmt`

For more information about this error, try `rustc --explain E0277`.                                                                                                                                                                                                                                                    
error: could not compile `smoltcp` due to previous error
warning: build failed, waiting for other jobs to finish...
```

Cargo.toml entry:
```
smoltcp = { version = "0.8.2", default-features = false, features = [
     "medium-ethernet", "medium-ip", "medium-ieee802154",
     "proto-ipv4", "proto-igmp", "proto-dhcpv4", "proto-ipv6",
     "socket-raw", "socket-icmp", "socket-udp", "socket-tcp", "socket-dhcpv4",
     "async", "defmt", "alloc"
] }
```

stevefan1999-personal, 2022-12-10:
Related: https://github.com/knurling-rs/defmt/issues/708

davidedellagiustina, 2022-12-22:
`defmt` has an `alloc` feature that should be enabled if you want `defmt::Format` to be implemented on structures from the `alloc` crate :)

Dirbaio, 2023-01-15:
Fixed in 9fa70d2781db7fec5836b80ad9b748726ef270a7 (#646)

closed 2023-01-15 by Dirbaio