smoltcp-rs/smoltcp #720: Possibility to support Stream Control Transmission Protocol (SCTP)?

overheat, 2022-12-21:
There is UDP implementation in this "TCP" stack. 
So any possibility to support  SCTP?

Dirbaio, 2022-12-22:
It's probably out of scope. It's very complex, and very rarely used in practice. What do you need it for?

overheat, 2022-12-22:
It's about media streaming, that we can see RTOS base device well support media streaming in the feature.

Sorry, I am not familiar with SCTP and don't know how complex it is. But I know some C language base code support it and can run in RTOS.

I assumed SCTP is beside with UDP and TCP. So we need another socket, SCTP socket?

overheat, 2022-12-22:
I check again and found that WebRTC data channel leverage SCTP. But looks like data channel feature is optional in WebRTC.

Dirbaio, 2022-12-22:
WebRTC uses SCTP over DTLS over UDP over IP. it's not the "real" SCTP, that would be SCTP over IP.

IMO WebRTC, SCTP, DTLS are out of scope of smoltcp (due to complexity, and of specificity of their uses). They would make more sense as separate crates, which could work on top of smoltcp's UdpSocket.

closed 2022-12-22 by overheat