smoltcp-rs/smoltcp #733: Make DHCP source and destination port configurable

Czocher, 2023-01-13:
Hello,
I have a small idea/request/problem. I have an embedded system which configures the IPv4 address by performing a DHCP request, but the request is performed on a different port than the standard one (eg. 1222 instead of 67, 1223 instead of 68). This is done not to interfere with existing DHCP servers which may be on the network.

Can this be configured somehow? If not, would you be willing to accept a PR which would add this kind of option?

Dirbaio, 2023-01-13:
This is not possible currently. 

The DHCP RFC mandates the use of ports 67/68, so changing it would be non-compliant. It seems a very strange thing to do to me. I see `dhclient` supports it, so there might be some uses cases, so adding it sounds good. It should be a `.set_ports()` method on the DhcpSocket probably.

Czocher, 2023-01-16:
Hello,
I provided a PR (#738) to support this use-case.

closed 2023-01-16 by Czocher