smoltcp-rs/smoltcp #778: consume function from TxToken return type

msmouni, 2023-04-21:
When using `Ethernet DMA` for sending packets, we cannot ensure that we are always able to send to packet by the time of calling the `consume` function:

https://github.com/smoltcp-rs/smoltcp/blob/0102a4b741655d7f6fd64110ceda5177580b3952/src/phy/mod.rs#L350-L361

Since the `DMA descriptors` can still be owned by the DMA when calling the `consume` function, we should be able to return an `Error` of type `WouldBlock` for example.

Dirbaio, 2023-04-21:
This is by design. You have to check for queue space in `receive` and `transmit`, return None if there's no space.

The reason for `consume` being infallible is: by the time `consume` is called, smoltcp has already done a ton of work to prepare the packet that goes to waste. it's better to signal failure as early as possible. It also makes the api simpler, just 1 way to fail instead of 2 (either `transmit` returning None, or returning Some and then `consume` failing).

This is definitely doable if the driver is structured correctly. I've done it in the [embassy-stm32 ethernet driver](https://github.com/embassy-rs/embassy/tree/master/embassy-stm32/src/eth), if you want to take a look at how.

closed 2023-04-21 by Dirbaio

msmouni, 2023-04-21:
Thanks for your answer, I understand better now that these traits are intended to be used in such a way that it assumes that one checks if the use of the `Token` will not lead to a failure before returning it through the `transmit` and `receive` functions.

This allows to invoke the `unwrap()` function on the Token results without having a panic:

https://github.com/embassy-rs/embassy/blob/master/embassy-stm32/src/eth/mod.rs#L120-L125