smoltcp-rs/smoltcp #780: What is the difference between using Loopback with Ethernet medium versus using Loopback with IP medium?

StevenJiang1110, 2023-04-24:
Now, when creating a Loopback device, it is necessary to specify the medium. I hope to send and receive TCP/UDP sockets on a Loopback device. Is there any difference in using these two mediums? The difference between these two mediums should be at the IP layer, but there should be no difference at the transport layer, right?

thvdveld, 2023-05-10:
When using the Ethernet medium, packets will be transmitted using Ethernet frames. It is then also required to let the interface know what the Ethernet address is. When using the IP medium, no link layer is used. The transport layer is untouched by the used medium.

StevenJiang1110, 2023-05-11:
Thanks a lot

closed 2023-05-11 by StevenJiang1110