smoltcp-rs/smoltcp #785: Is `smoltcp::socket::dns` only DNS client or also DNS server?

vi, 2023-05-10:
`smoltcp::socket::dns` has no module-level documentation.

`smoltcp::socket::dns::Socket`'s documentation ("A Domain Name System socket. A UDP socket is bound to a specific endpoint, and owns transmit and receive packet buffers.") also does not make clear which features are supported and which are not.

* Can it do recursive DNS lookup?
* Can it used for implementing a DNS server, not for using DNS server?
* Can it decode captured DNS query and/or answer (not the one created by `smoltcp::socket::dns`)?

Dirbaio, 2023-05-10:
> Can it do recursive DNS lookup?

No. 

> Can it used for implementing a DNS server, not for using DNS server?

No, client only.

> Can it decode captured DNS query and/or answer (not the one created by smoltcp::socket::dns)?

Not sure what you mean. It parses the answer, you can obtain the IP address list from it with this function https://docs.rs/smoltcp/latest/smoltcp/socket/dns/struct.Socket.html#method.get_query_result

vi, 2023-05-10:
Thanks for the answer. This should be in the docs.

I expected there to be something like [dns_parser::Packet::parse](https://docs.rs/dns-parser/0.8.0/dns_parser/struct.Packet.html#method.parse). Smoltcp's analogue seems to be `smoltcp::wire::DnsPacket`, though it exposes only basic things. Existence of DNS-related code on wire level can also acknowledged in `smoltcp::socket::dns` docs, as this module is the first search result when looking up "dns".