smoltcp-rs/smoltcp #818: Contributing - is project still active?

victor-wei126, 2023-07-19:
I'd like to possibly contribute to this project - is it still actively maintained?

crawford, 2023-07-19:
Yep. Send over your patches. There are a number of folks who will review them.

victor-wei126, 2023-07-20:
Great! I'll take a look at the issues and start work on one - any particularly pressing issues?

pothos, 2023-07-21:
Not pressing but nice to have for reliable operation are the "not implemented", "not supported" parts of TCP: https://github.com/smoltcp-rs/smoltcp#tcp-layer
Silly window syndrome avoidance and Probing Zero Windows are both not too large in scope but require some RFC reading and thinking. Selective acknowledgements is probably a bit larger.
Someone even started congestion control, not sure where the old branch is but you could also take one of the BBR implementations from a Go network stack for comparison.

closed 2023-08-17 by oslfmt