smoltcp-rs/smoltcp #820: phy::RawSocket incompatible with Medium::Ip

toshipiazza, 2023-07-25:
`RawSocketDesc` creates an AF_PACKET socket with libc::SOCK_RAW in case of any medium

https://github.com/smoltcp-rs/smoltcp/blob/fa7fd3c321b8a3bbe1a8a4ee2ee5dc1b63231d6b/src/phy/sys/raw_socket.rs#L21-L40

However, this doesn't work when Medium::Ip is passed to phy::RawSocket and to RawSocketDesc, since the incoming packets have an L2 frame prepended but Interface expects packets to start from the IP header

According to PACKET(7) man page

> The  socket_type  is either SOCK_RAW for raw packets including the link-level header or SOCK_DGRAM for cooked packets with the link-level header removed.  

For Medium::Ip we should set socket_type = libc::SOCK_DGRAM | libc::SOCK_NONBLOCK and protocol = libc::ETH_P_IP so the resulting packets start with the IP header

toshipiazza, 2023-07-25:
I found that I also needed to change the libc::send() to libc::sendto() and provide an libc::sockaddr_ll structure in RawSocketDesc::send()

toshipiazza, 2023-07-25:
I guess it's unclear if this is expected or desired to work, since the docs for Medium::Ip suggest that Medium::Ip is an abstraction designed for TUN devices...

If this is not intended to work it may make sense to assert_ne!(medium, Medium::Ip)

Dirbaio, 2023-07-25:
yes, it's intended to work. Can you send a PR?

toshipiazza, 2023-07-27:
This may be harder to fix than how I initially thought. When sending a packet on a socket created with AF_PACKET + SOCK_DGRAM it looks like you need to specify the L2 destination address in a struct sockaddr_ll to sendto(). This doesn't work well with the Medium::Ip abstraction/design