smoltcp-rs/smoltcp #852: Can multiple tcp sockets listening on the same endpoint of one interface?

StevenJiang1110, 2023-10-19:
In linux, the listen syscall can accept a `backlog` parameter, so it can accept multiple connection request meantime.

Since the listen method in smoltcp does not has a backlog paramter now, I wonder if I can construct multiple socket  listening on the same address to mimic the similar behavior?

I have tried it but the second connection will reports error that ACK cannot be accepted. So is it impossible by design? or is it only my programming mistake?

Dirbaio, 2023-10-19:
There's no equivalent of linux "backlog", each socket can accept only 1 connection. If you want to accept multiple, create N sockets, call listen() on them on the same port -- it's fine to have N listening sockets on the same port, incoming connections go to a random one.

StevenJiang1110, 2023-10-20:
Thanks a lot. The problem is solved.

closed 2023-10-20 by StevenJiang1110