smoltcp-rs/smoltcp #856: plan of support TunTapInterface on macos?

ibigbug, 2023-11-14:
there are some libraries supporting it, such as https://github.com/meh/rust-tun/blob/master/src/platform/macos/device.rs

any plans of doing so in smoltcp?

thvdveld, 2023-11-28:
I'm totally not an expert in TUN/TAP interfaces, but maybe smoltcp already supports this in https://github.com/smoltcp-rs/smoltcp/blob/main/src/phy/tuntap_interface.rs? And if not, I don't think it is planned. But feel free to open a PR to add support.

ibigbug, 2023-12-31:
thanks - i think when it goes to https://github.com/smoltcp-rs/smoltcp/tree/main/src/phy/sys , only Linux is supported 

I'll try to add this after I understand how to do it

hainesc, 2024-04-09:
@ibigbug just kind ping to know what is the process now, because I got here too. 

Maybe windows support with wintun together is a good idea.