smoltcp-rs/smoltcp #903: UDP: Read and set local address on unbound sockets (IPV6_PKTINFO style functionality)

chrysn, 2024-02-02:
When UDP sockets are created and bound to a port but no IP address (the POSIX equivalent being being bound to `[::]:${PORT}`), applications often need to know which local address the packet was sent to, so that responses can be sent from the same address. This is requirement for implementing CoAP and (AIU) also QUIC -- and part of the embedded-nal-async (where [more extensive rationale](https://github.com/rust-embedded-community/embedded-nal/pull/73#issuecomment-1262041739) is given).

I can put in work on smoltcp to make this possible, but being new to the project I'll need a bit of guidance as to where to do that. My naive approach would be to add a field `localendpoint` to PacketMetadata, possibly behind a `pktinfo`ish flag. Odd thing is that this is really not phy-level information -- it'd just as well fit in `UdpMetadata`, but that is not `#[non_exhaustive]` (should it be?).

For actually making it functional on the incoming side, I think I'd just need to update socket::udp::Socket::process to propagate data that is available in the ip_repr down into the metadata. On the send side, this would be in its `dispatch()`, where we'd have to decide whether we just take the application's word that the source address is good, or whether we check if it really matches self.endpoint.addr / is a valid source address of cx.

Dirbaio, 2024-02-02:
yes, UdpMetadata would be the correct place to add this.

chrysn, 2024-02-02:
So, I add it to UdpMetadata in a PR and worry about non_exhaustive and semver breakage later? Should I still make it opt-in, or make it unconditional? (Any implementation of embedded-nal-async on it, such as the one I'd plan for embassy, would enable the flag anyway, and it's about a 128bit copy and per-message storage that's added, maybe even saving on get_source_address calls in the outbound direction.)

Dirbaio, 2024-02-02:
don't worry about breakage, next release will likely be 0.12 (breaking bump) anyway.

I'm not sure about making it optional.. it's 16 extra bytes per packet, which is not much but also not insignificant. i'm sort of leaning towards not making it optional, because we already have enough feature flags :smiling_face_with_tear: