smoltcp-rs/smoltcp #908: Use of `fugit` instead of custom time implementation.

liamkinne, 2024-02-19:
Would the maintainers be open to using [`fugit`](https://docs.rs/fugit/latest/fugit/) for the `Instant `and `Duration` rather than the custom implementation within this project?

Would save a bit of complexity in my code having to convert between the smoltcp types and the fugit types used by the SysTick library I'm using.

If someone gives the go ahead, I'm happy start working on a PR.

Dirbaio, 2024-02-19:
I personally think `fugit` is a bit hard to use with the u32/u64/etc and NUM/DENOM generics. 

besides, what timebase (NUM/DENOM) would smoltcp use for the time? still hardcode it to microseconds? in that case you'll still have to convert between that and whatever your microcontroller is using.

liamkinne, 2024-02-19:
Yeah I did think about the timebase problem. Fugit does have a method [`.convert()`](https://docs.rs/fugit/latest/fugit/struct.Duration.html#method.convert) to handle different time bases. Given conversion is built-in, then I think a microsecond timebase would be fine and doesn't deviate from the current implementation.

Fugit makes the most sense to _me_ because of it's [ubiquity](https://crates.io/crates/fugit/reverse_dependencies) among HAL implementations and so in my mind having support for at least them is better than for none.

Other options are feature-gating the use of fugit or trying to make the time provider generic (my least favorite option to be clear).