smoltcp-rs/smoltcp #923: How can I get neighbour cache from `Interface`?

ruiqurm, 2024-04-19:
Hello, I'm new to `smoltcp` and am trying to retrieve a MAC address from a given IP address. It appears there are no direct methods in smoltcp for querying MAC addresses. However, I'm considering an indirect approach. I understand that sending any packet prompts a MAC address lookup, which means I could theoretically send a random UDP packet to cache the MAC address. But, how can I then access this cached MAC address? I've reviewed the documentation for the [Interface](https://docs.rs/smoltcp/latest/smoltcp/iface/struct.Interface.html) but couldn't find a method to do this. Any suggestions or guidance would be greatly appreciated. Thank you!

thvdveld, 2024-05-23:
What is your use case? Your solution seems to be very convoluted. Wouldn't it be easier if the Interface could return an iterator over the neighbors it has in its cache?

ruiqurm, 2024-05-23:
> What is your use case? Your solution seems to be very convoluted. Wouldn't it be easier if the Interface could return an iterator over the neighbors it has in its cache?

I am developing a user-space RDMA driver for custom hardware. Currently, the hardware does not support sending and replying to ARP messages automatically. Instead, the software is provided with the IP address and is responsible for finding the corresponding MAC address. The software can send raw packets to the hardware, which then transmits them directly to the network. I use smoltcp to construct ARP queries and responses to obtain MAC addresses.

However, there seems to be no public interface to retrieve the neighbor cache directly. For various reasons, I haven't modified the API and am relying on my version. Alternatively, I tried to use the current public API. My current workaround is to send an ICMP packet to the other side and check and store the MAC address when handling the TX token, which is quite convoluted.

I am willing to implement an API that allows querying the MAC address from a given IP if someone can help review the code.

thvdveld, 2024-05-23:
> I am willing to implement an API that allows querying the MAC address from a given IP if someone can help review the code.

Do you mean querying a given IP locally on the device or remotely using some protocol? I'm not familiar with RDMAs. If it is locally, you could use the following function if we make it public.

https://github.com/smoltcp-rs/smoltcp/blob/main/src/iface/interface/mod.rs#L888C1-L901C6

ruiqurm, 2024-05-23:
> > I am willing to implement an API that allows querying the MAC address from a given IP if someone can help review the code.
> 
> Do you mean querying a given IP locally on the device or remotely using some protocol? I'm not familiar with RDMAs. If it is locally, you could use the following function if we make it public and modify it to return an Option.
> 
> https://github.com/smoltcp-rs/smoltcp/blob/main/src/iface/interface/mod.rs#L888C1-L901C6

Just query a given IP locally.  What I need is a combination of [has_neighbor](https://github.com/smoltcp-rs/smoltcp/blob/ef67e7b46cabf49783053cbf68d8671ed97ff8d4/src/iface/interface/mod.rs#L889) and [lookup_hardware_addr](https://github.com/smoltcp-rs/smoltcp/blob/ef67e7b46cabf49783053cbf68d8671ed97ff8d4/src/iface/interface/mod.rs#L904). When I call the method, it should first query the neighbor cache. If the cache misses, it should send an ARP query automatically and return None.

It might be better to provide a blocking version where the user can set a timeout and wait for the query result.

ruiqurm, 2024-05-24:
> Do you mean querying a given IP locally on the device or remotely using some protocol? I'm not familiar with RDMAs. If it is locally, you could use the following function if we make it public and [...]

Hi. It is not simple to query the Mac address through just one interface. Maybe I can write an ARP socket so that users can query arp manually?