smoltcp-rs/smoltcp #927: [Feature Request] Resize TCP socket buffer size

XOR-op, 2024-05-12:
Currently, the buffer for TCP socket needs to be preallocated in `socket::tcp::Socket::new()` and I found nowhere to change its size. However, a configurable window size is necessary to achieve a balance between good bandwidth and low memory footprint. I notice that TCP window scaling (RFC 1323) is advertised to be implemented but it seems it's only properly implemented for the remote endpoint. Can this feature be implemented in local side?

Dirbaio, 2024-05-12:
I'm not sure what feature you're requesting:

- You can already use buffers any size you want. If you want a bigger buffer (hence a bigger window), use a bigger buffer.
- TCP window scaling is fully implemented. if you set a buffer bigger than 65536, window scaling will be used to advertise the full buffer size to the remote side. (Also note window scaling is just a TCP extension allowing advertising window sizes larger than 16 bits, which is the window field size in the header. Window scaling by itself won't make the buffer bigger.)

XOR-op, 2024-05-12:
Thanks for your replying. What I want to achieve is the following:
- I don't know if the connection needs a big bandwidth initially, so I create a TCP socket with `tcp::Socket::new(vec![0;65536], vec![0; 65536])` to reduce memory usage. 
- After some data exchange, I notice that we need a bigger receive window to satisfy BDP. Then I want to make the receive window to be 1MB. (Always allocating 1MB buffer for every socket at the beginning will lead to much bigger memory footprint)

Is this already possible?

Dirbaio, 2024-05-12:
ah okay! you want to dynamically *change* the buffer size after the TCP socket has already been created. Makes sense.

Unfortunately that's not possible now, no. Doing it if the buffer is borrowed might be hard, but should be doable if it's an owned Vec.

XOR-op, 2024-05-12:
Then can we add such an API for sockets with owned Vec or `Socket<'static>` such as
```rust
fn replace_recv_buf<T: Into<SocketBuffer<'static>>(&mut self, new_buffer: T)->Option<SocketBuffer<'static>>
```
? Enabling this feature will be helpful to memory-constraint devices with skewed bandwidth needs.

croepha, 2026-09-02:
Would it be low hanging fruit to at-least release the buffer for sockets in a TIME_WAIT state? I've got some high connection churn tests where this is the biggest bottleneck.  I have a connection limit in-order to keep from hitting OOM.

Even if it weren't automatically dynamic, if it were possible to remove a connection, then re-add it with different buffers (retaining the TCP state) then that could work as a workaround.

So... in my case, I have to set a really high RX buf size because there is high latency.  I don't actually need smoltcp to hold onto the data, I just need it to advertise a high window to the peer...  I am pulling the data out of smoltcp very quickly, so the large RX buffer is kinda superficial...