smoltcp-rs/smoltcp #932: Smoltcp panics in `get_source_address_ipv6`

dscso, 2024-05-20:
**TL;DR:**

I think I am having the same issue as #894, maybe creating a new release could fix it.

My whole live story:

I am sorry that I can not provide much more details, but I am using an ESP32 with `embassy-net` v0.4.0 (which uses internally  `smoltcp` v0.11.0). My program is not crazy complex, it gets an IPv4 address using DHCP, and for testing I am binding to some sockets and do some DNS queries.

I have connected the ESP32 to a network with a looot of "smart" devices. On this network the ESP32 crashes in random times with the error pasted below. This crash occurs within a few dozen seconds to a few minutes. When I put the ESP32 in an isolated network without the smart devices, there seem to be no issue. I left it running over night and it worked just fine.

I tried to compile the network stack without IPv6 enabled to check if the error also occurs when IPv6 is disabled unfortunately, IPv6 seems not so easy to disable using the `embassy-net` crate.

```
!! A panic occured in '/Users/me/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/iface/interface/mod.rs', at line 882, column 14:
called `Option::unwrap()` on a `None` value

Backtrace:

0x400e3ce1
0x400e3ce1 - core::option::unwrap_failed
    at /Users/me/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/option.rs:1978
0x400fadcc
0x400fadcc - core::option::Option<T>::unwrap
    at /Users/me/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/option.rs:931
0x400f71c6
0x400f71c6 - smoltcp::iface::interface::ipv6::<impl smoltcp::iface::interface::InterfaceInner>::icmpv6_reply
    at /Users/me/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/iface/interface/ipv6.rs:336
0x400f7e4d
0x400f7e4d - smoltcp::iface::interface::InterfaceInner::process_udp
    at /Users/me/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.11.0/src/iface/interface/mod.rs:1174
0x400f2e04
0x400f2e04 - smoltcp::iface::interface::ipv6::<impl smoltcp::iface::interface::InterfaceInner>::process_nxt_hdr
    at ??:??
0x400dd987
0x400dd987 - smoltcp::iface::interface::ethernet::<impl smoltcp::iface::interface::InterfaceInner>::process_ethernet
    at ??:??
0x400d3b6f
0x400d3b6f - embassy_net::Inner<D>::poll
    at /Users/me/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-net-0.4.0/src/lib.rs:857
0x400da04f
0x400da04f - embassy_executor::raw::SyncExecutor::poll::{{closure}}
    at /Users/me/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.5.0/src/raw/mod.rs:405
0x400e294f
0x400e294f - esp32_rust_mqtt::__xtensa_lx_rt_main
    at /Users/me/code/mqtt/esp32-rust-mqtt/src/main.rs:29
0x400ff592
0x400ff592 - Reset
    at /Users/me/cargo/registry/src/index.crates.io-6f17d22bba15001f/xtensa-lx-rt-0.16.0/src/lib.rs:70
```

thvdveld, 2024-05-23:
This issue should be fixed with #895. Could you try using the main branch of smoltcp and see if that fixes your issue?

dscso, 2024-05-23:
Yes I tried it now and it seems to be fixing the issue, but I have to leave it running a bit longer. Interestingly now the binary size decreased 🤔...

What would be the steps to create a new release so it could be used in embassy-net?

Dirbaio, 2024-05-24:
fixed in https://github.com/smoltcp-rs/smoltcp/pull/895.

closed 2024-05-24 by Dirbaio