smoltcp-rs/smoltcp #981: Difference between a ListenEndpoint with addr == None and addr == Some(Ipv4Address::UNSPECIFIED)

kbleeke, 2024-08-30:
I've run into a problem where UDP sockets created with `bind(8080)` [port is arbitrary] and `bind(Ipv4Addr::UNSPECIFIED, 8080)` behave differently. 

The first behaves as expected: it accepts packet from any source IP and sends packets with the local IP address in the source field.

The second, however, seems (to me unexpectedly) take Ipv4Addr::UNSPECIFIED literally and does not accepts packets from any address. It also sends packet with 0.0.0.0 as the source address instead of substituting the correct IP. I was able to receive packets from the broadcast address, presumably it also accepts packet with 0.0.0.0 as the source address.

I am not using smoltcp directly but embassy_net but the relvant code here seems to be in smoltcp.

Dirbaio, 2024-08-30:
This is intended behavior, smoltcp represents some IP as Some, and no IP as None. `Some(0.0.0.0)` means exactly that, the `0.0.0.0` IP (which is invalid).

using Some/None is idiomatic Rust, using the zero value in-band to signal "no value" is not, it's a C thing.

kbleeke, 2024-09-02:
Ok, thanks. Nothing to be done here, then

/close

closed 2024-09-02 by kbleeke