smoltcp-rs/smoltcp #999: How we are using `smoltcp`

thomaseizinger, 2024-10-16:
I reached for `smoltcp` because I needed to implement a TCP DNS server inside a TUN driver. The code is here: https://github.com/firezone/firezone/tree/main/rust/dns-over-tcp/src

Because we are also handling all sorts of other traffic and `smoltcp` is only responsible for managing TCP sockets, I had to create some adapter structs and workarounds to be able to selectively dispatch IP packets to it. It actually works pretty well but it took me a while to figure these things out.

- Is there any interest having these adapters in this repository in case other people have similar usecases?
- Is there maybe a better way of doing this and I haven't noticed?
- Should there be a better way of using `smoltcp` if you already have IP packets and want to just give it to `smoltcp` for processing?

thvdveld, 2024-11-28:
> - Is there any interest having these adapters in this repository in case other people have similar usecases?

I'm not sure what you mean with these adapters. Could you point to the source more specific?

> - Should there be a better way of using smoltcp if you already have IP packets and want to just give it to smoltcp for processing?

Currently, there is no public function that just handles one IP packet. Interacting with smoltcp is only done using the poll functions. This is because smoltcp does more than just handling a packet. For example, it needs to update fragmentation buffers, it needs to clear outdated neighbours. We have private functions that process single packets, but making them public will be a foot gun.

thomaseizinger, 2024-11-28:
> > * Is there any interest having these adapters in this repository in case other people have similar usecases?
> 
> I'm not sure what you mean with these adapters. Could you point to the source more specific?

Sure.

Here is the `InMemoryDevice`: https://github.com/firezone/firezone/blob/main/rust/dns-over-tcp/src/stub_device.rs

Here is the accompanying `Interface`: https://github.com/firezone/firezone/blob/main/rust/dns-over-tcp/src/interface.rs

Here is how it is initialised: https://github.com/firezone/firezone/blob/3a62709c77931644e33f5c76f5c9f93b1fbaa7f4/rust/dns-over-tcp/src/server.rs#L53

And how we feed packets to it: https://github.com/firezone/firezone/blob/3a62709c77931644e33f5c76f5c9f93b1fbaa7f4/rust/dns-over-tcp/src/server.rs#L141

Overall, the current API feels a bit "object-oriented". I need to create a kind of stub-device that I feed with a buffer because `Interface` depends on the "token" abstraction of `Device`. Ultimately, I think this could be decoupled: `Interface` needs an API that accepts packets and emits packets. Where does come from doesn't matter. The `poll`-interface already necessitates the creation of an event-loop. It would add much complexity if users would have to poll the device directly and pass IP packets to `Interface` instead of having `Interface` depend on the device abstraction.