smoltcp-rs/smoltcp #1006: Version 0.11.0 does not compile

n0x32schneider, 2024-10-26:
Hello,
I want to use smoltcp in a no_std environment, but version 0.11.0 does not compile.

Version 0.10.0 compiles without trouble in the same empty code skeleton (cortex-m-quickstart repository).

Dependency entry in Cargo.toml looks like

```
[dependencies]
cortex-m = "0.6.0"
cortex-m-rt = "0.6.10"
cortex-m-semihosting = "0.3.3"
panic-halt = "0.2.0"
smoltcp = { version = "0.11.0", default-features = false, features = ["log", "medium-ethernet", "proto-ipv4", "socket-tcp"] }
```

Using version Rustc 1.82.0 (f6e511eec 2024-10-15)

Anyone else experiencing this, or is it just me?

thvdveld, 2024-10-27:
I think it is because you enabled the `log` feature-flag for smoltcp, which does not compile for no_std. Instead, you could enable the `defmt` feature flag to enable logging, using the defmt crate.

n0x32schneider, 2024-10-27:
Thanks for the response! Compiling with the ```log``` feature-flag does seem to compile anyways.

The problem is resolved, for some reason cargo had stored a broken version of smoltcp 0.11.0 in it's local registry.
Removing it fixed the issue!

closed 2024-10-27 by n0x32schneider