smoltcp-rs/smoltcp #1019: TUN inerface is down when using TCP raw socket

alishir, 2024-12-16:
Hi,
I tried to create a TCP raw socket with a TUN interface using the following snippet. The `tun0` device was created successfully; however, it is currently down. Unfortunately, I could not find any function to bring the interface up. Do you know how I can bring the interface up?

```rust

use smoltcp::socket::raw::PacketMetadata;
use smoltcp::wire::Ipv4Address;
use smoltcp::wire::{IpAddress, IpCidr};
use smoltcp::{
    iface::{Config, Interface, SocketSet},
    phy::TunTapInterface,
    socket::raw,
    storage::PacketBuffer,
    time::Instant,
    wire::{Ipv4Packet, PrettyPrinter},
};

fn main() {
    let mut tun_device = TunTapInterface::new("tun0", smoltcp::phy::Medium::Ip).unwrap();

    let mut config = Config::new(smoltcp::wire::HardwareAddress::Ip);
    config.random_seed = rand::random();

    let mut iface = Interface::new(config, &mut tun_device, Instant::now());
    iface.set_any_ip(false);
    iface.update_ip_addrs(|ip_addrs| {
        ip_addrs
            .push(IpCidr::new(IpAddress::v4(192, 168, 69, 1), 24))
            .unwrap();
    });
    iface
        .routes_mut()
        .add_default_ipv4_route(Ipv4Address::new(192, 168, 69, 100))
        .unwrap();

    let rx_buffer = PacketBuffer::new(vec![PacketMetadata::EMPTY; 1500], vec![0; 1500]);
    let tx_buffer = PacketBuffer::new(vec![PacketMetadata::EMPTY; 1500], vec![0; 1500]);
    let raw_socket = raw::Socket::new(
        smoltcp::wire::IpVersion::Ipv4,
        smoltcp::wire::IpProtocol::Tcp,
        rx_buffer,
        tx_buffer,
    );

    let mut sockets = SocketSet::new(vec![]);
    let raw_handle = sockets.add(raw_socket);
    loop {
        iface.poll(Instant::now(), &mut tun_device, &mut sockets);
        let socket = sockets.get_mut::<raw::Socket>(raw_handle);
        if socket.can_recv() {
            let buffer = socket.recv().unwrap();
            println!("{}", PrettyPrinter::<Ipv4Packet<&[u8]>>::new("", &buffer));
        }
    }
}

```

Dirbaio, 2024-12-16:
you have to configure it and bring it up yourself from the linux side. readme has instructions https://github.com/smoltcp-rs/smoltcp?tab=readme-ov-file#hosted-usage-examples (it works the same for tun and tap)

alishir, 2024-12-17:
Thanks a lot. Since I see that the TUN device creation, IP, and route assignments were done via some Rust code in the examples, I think it is possible to bring the device up from Rust.

closed 2024-12-17 by alishir