smoltcp-rs/smoltcp #1030: ARP requests never succeed (timeout exceeded)

dvdsk, 2025-01-05:
Stack: Embassy commit a4f8fdd, using the embassy-net-wiznet  crate (0.1).

Context: 
this happens after about 15 hours of working perfectly though that period is not fixed.

What I need:
Some help debugging this. Is there a place in smolltcp I should add extra tracing? I tried some already but I did not get wiser from it. I might also need some reading up on ARP etc, forgotten most of that stuff now. Is there a resource you can recommend?

source snippet:
```rust
pub async fn handle(
    stack: &Stack<'_>,
    publish: &Queues,
    led: LedHandle<'_>,
    driver_orderers: &slow::DriverOrderers,
) {
    let mut rx_buffer = [0; 1024];
    let mut tx_buffer =
        [0; max(SensMsg::ENCODED_SIZE, ErrorReport::ENCODED_SIZE) * 2];

    let mut socket = TcpSocket::new(*stack, &mut rx_buffer, &mut tx_buffer);
    socket.set_timeout(Some(Duration::from_secs(5)));
    socket.set_keep_alive(Some(Duration::from_secs(1)));
    let host_addr = Ipv4Address::new(192, 168, 1, 43);
    let host_port = 1234;

    debug!("Configured socket and connecting");
    loop {
        debug!("socket state: {:?}", socket.state());
        if let Err(e) = socket.connect((host_addr, host_port)).await {
            warn!("connect error: {}", e);
            Timer::after_secs(5).await;
            continue;
        }

        info!("(re-)connected");
        // Prevent out-dated data from being send
        publish.clear().await;

        let (reader, writer) = socket.split();
        match select(
            send_messages(writer, publish),
            receive_orders(reader, &led, &driver_orderers),
        )
        .await
        {
            select::Either::First(e) => {
                warn!("Error while sending messages: {}", e)
            }
            select::Either::Second(e) => warn!("Error receiving orders: {}", e),
        };
        // Or the socket will hang for a while waiting to close this makes sure
        // we can reconnect instantly
        socket.abort();
        Timer::after_secs(60).await; // Experiment: does this help?
    }
}
```

`send_messages` and `receive_orders` are rather simple. The former using `socket.write_all()` then `socket.flush()` the latter only does `socket.read()`.

trace:
(note I edited the paths `/home/david/.cargo/registry/src/index.crates.io-6f17d22bba15001f/smoltcp-0.12.0/src/macros.rs:18` becomes `smoltcp/src/macros.rs:18`
```rust
WARN  connect error: ConnectionReset
└─ large_bedroom_sensor_node::network::handle::{async_fn#0} @ src/network.rs:103
DEBUG timeout exceeded
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG timeout exceeded
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG timeout exceeded
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG timeout exceeded
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG timeout exceeded
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG socket state: Closed
└─ large_bedroom_sensor_node::network::handle::{async_fn#0} @ src/network.rs:95
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
DEBUG timeout exceeded
└─ smoltcp::socket::tcp::{impl#9}::dispatch @ smolltcp/src/macros.rs:18
DEBUG address 192.168.1.43 not in neighbor cache, sending ARP request
└─ smoltcp::iface::interface::{impl#2}::lookup_hardware_addr @ smolltcp/src/macros.rs:18
WARN  connect error: ConnectionReset
```

Dirbaio, 2025-01-05:
this is more likely a bug in `embassy-net-wiznet`. Do you have a packet capture from the other host that would show whether the ARP requests are coming through or not?

dvdsk, 2025-01-14:
> Do you have a packet capture from the other host that would show whether the ARP requests are coming through or not?

Now I have.

Arp requests are coming through fine until the issue occurs. Then no more frames arrive at all from the device until it is reset. After reset arp works again and normal traffic resumes until the issue occurs again. Do you need the package capture or should I open an issue at embassy?

Dirbaio, 2025-01-14:
yeah this seems an embassy-net-wiznet issue. please open an issue there. Be warned these "stops working randomly after many hours" issues are super tricky to debug, if you want other people to help you should post a fully self-contained repro other people can run on some off-the-shelf board, ideally one of the wiznet eval boards.