smoltcp-rs/smoltcp #1037: Binding to "any" address does not work as expected

chrysn, 2025-01-27:
At least on UDP sockets, calling `socket.bind(IpEndpoint::from(core::net::SocketAddr::parse("0.0.0.0:1234").unwrap()))` does not behave as expected: It seems to bind not at all, whereas it should bind to that port on any IP address.

There is a workaround of doing `socket.bind(1234)` -- that's fine when an application calls it directly, but needs a workaround such as

```
        let bind_to = IpEndpoint::from(local);
        if bind_to.addr.is_unspecified() {
            socket.bind(bind_to.port)?;
        } else {
            socket.bind(bind_to)?;
        }
```

(And sorry for the IPv4 examples, we're working to have v6 [in parallel](https://github.com/smoltcp-rs/smoltcp/issues/570#issuecomment-2497546647))

I can most probably provide a PR later; right now, this serves to document the workaround.

Dirbaio, 2025-01-27:
this is intended behavior, see #981

chrysn, 2025-01-27:
Fine with me, sorry for the duplicate.

That's an argument that will also be helpful in <https://github.com/rust-embedded-community/embedded-nal/issues/92> -- indicating that the core types are a tad too far on the "POSIX sockets" side of things, with more idiomatic expressions around (esp. when the IP addresses are expressed in such a way that they have a zero niche).

closed 2025-01-27 by chrysn