smoltcp-rs/smoltcp #1048: TcpSocket panicked on "attempt to subtract sequence numbers with underflow"

zonyitoo, 2025-03-25:
```
thread 'smoltcp-poll' panicked at /Users/zonyitoo/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/smoltcp-0.12.0/src/wire/tcp.rs:81:13:
attempt to subtract sequence numbers with underflow
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Why the sequence number underflowed?

caobug, 2025-06-16:
Did you solve it? I had the same issue and had to remove the panic! macro.

https://github.com/caobug/smoltcp/commit/b3fc74cbb05f0b1951cf8211a4051dd93ba043d9

zonyitoo, 2025-06-17:
It couldn't be reproduced after I disabled the RX checksum.

zonyitoo, 2025-07-26:
Related issue: https://github.com/smoltcp-rs/smoltcp/issues/1051

zonyitoo, 2025-07-26:
`last_scaled_window` was introduced by this commit: https://github.com/smoltcp-rs/smoltcp/commit/fe179c918415d6c251384b24fdd6204060265431 . @lrh2000 Please take a look at this issue.

lrh2000, 2025-07-28:
I think that it is theoretically possible to receive more bytes than the previously advertised window:
https://github.com/smoltcp-rs/smoltcp/blob/d2d647090d544b1e7c142571da9d55f7280f664b/src/socket/tcp.rs#L1598-L1599

If the receivers consumes some bytes to free some space in `self.rx_buffer` and a misbehaved sender tries to produce more bytes than our previously advertised window, we may accept that as long as the RX buffer has enough space.

In that case, `last_ack + last_win < next_ack` seems possible to happen.

zonyitoo, 2025-08-06:
@lrh2000 Could you make a fix PR for this?