smoltcp-rs/smoltcp #1056: core::net::ip_addr::Ipv6Addr::Display (and other Display implementations) bloating image sizes

xkentr, 2025-04-24:
Is there any way to eliminate the usage of Ipv6Addr? As soon as I use it, it adds 2KiB to my image size for an embedded target.

` 0.8%   3.1%  2.0KiB          std <core::net::ip_addr::Ipv6Addr as core::fmt::Display>::fmt`

There are other Display implementations that are relatively large too, like icmpv6::Message. With only Ipv6 icmp enabled, approximately 10KiB is fmt::Display implementations, but I have no output device with which to display them.

xkentr, 2025-04-24:
Update: I have tracked this down to a handful of "panic!" macros that format errors including large "fmt" implementations.

Would a pull request to simplify these panics to simple string messages be accepted?

Dirbaio, 2025-04-24:
If youŕe getting `core::fmt` stuff imported into your binary, it's because you have something pulling it in. For example you might have a panic handler that is actually formatting the `PanicInfo`. If the panic handler ignores it, most of the `fmt` stuff gets optimized out because it's unused.

Also check you're building with `--release`, and try adding this to Cargo.toml to enable LTO.

```toml
[profile.release]
codegen-units = 1
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 'z'
overflow-checks = false
```

This should optimize most of the bloat out. There might be a few panic strings left. One trick that works to remove the last bloat is building with nightly and adding this to `.cargo/config.toml`

```toml
[unstable]
build-std = ["core"]
build-std-features = ["panic_immediate_abort"]
```

We shouldn't simplify the panics in smoltcp itself because they are useful when debugging when you do have logging output. You can always remove the `fmt` bloat if you set up your project with the right panic handler and optimizations when you don't have/want logging output.

xkentr, 2025-04-24:
Yes, I have lto and opt-level = "s" turned on, and took care to avoid using any string formatting in my code. I'm pretty sure the panic invocations are the contributors because if I edit them to remove the formatting the Display::fmt code the resulting executable no longer has that compiled code in it.

The suggestion to use
```
[unstable]
build-std = ["core"]
build-std-features = ["panic_immediate_abort"]
```

worked well. This eliminated all Display::fmt code from any source and all of the core fmt functions as well, the resulting executable  was 13KiB smaller. Thanks!

closed 2025-04-24 by xkentr